EdgeTX/edgetx #6178: a potentially stupid feature request but a B&W mode for the colour touch screens

MrBlackmidi, 2025-05-01:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

As the title suggests, this may be a stupid idea, but a feature to have a black and white mode for the colour touch screen (which disables the touch screen and simulates the monochrome lcd screens like the zorro or the boxer)

### Describe the solution you'd like

Either have the text to be dot matrix like the monochrome LCDs or the full resolution of the colour screen. a bonus would be to allow the user to change the colour of the "screen" too

### Describe alternatives you've considered

there isn't one afaik

### Additional context

[image]

philmoz, 2025-05-01:
We already have to support two B&W formats, and 3 Color formats (with two more on the way).

With the limited resources available it is unlikely this will ever happen.

gagarinlg, 2025-05-01:
Sounds like you get the worst of two worlds

the unintuitive B&W UI + the power consumption of a big Color TFT LCD

pfeerick, 2025-05-01:
I'm afraid I have to agree... while I do think it would be a novel idea (and would result in a single, and thus consistent UI, we simple don't have the resources to do this. The nearest you will likely get is to create a B&W  theme...

closed 2025-05-01 by pfeerick

3djc, 2025-05-01:
Would have been back to Otx, since 'some' users do prefer the higher density of informations that Otx was able to offer on a color lcd, at the expense of no touch support.