EdgeTX/edgetx #6181: EdgeTx simulator: No radio profiles has been found. Use EdgeTx Companion to create

snowuyl, 2025-05-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I run simulator211 with the following command. But EdgeTx simulator shows "No radio profiles has been found. Use EdgeTx Companion to create".
edgetx_simulator/build/native/simulator211 

### Expected Behavior

Running simulator211 successfully.

### Steps To Reproduce

Please following the following command to build EdgeTX simulator.
sudo apt update
sudo apt install libstdc++6

pip3 install clang
pip3 install jinja2
pip3 install lz4
pip3 install pillow

./tools/setup_buildenv_ubuntu20.04.sh
git clone --recursive -b main https://github.com/EdgeTX/edgetx.git edgetx_simulator
cd edgetx_simulator
git checkout 2.11
git branch
qmake --version

mkdir build
cd build
cmake -DARM_TOOLCHAIN_DIR=/opt/gcc-arm-none-eabi/bin -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make configure
make -j`nproc` libsimulator
make -j`nproc` companion
make -j`nproc` simulator

Please using the following command to run simulator211.
edgetx_simulator/build/native/simulator211 

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Linux

### OS Version

Ubuntu 20.04

snowuyl, 2025-05-02:
[

[image]

](url)

pfeerick, 2025-05-02:
Literally what it says to do... Run Companion first. Create a radio profile (name it, pick the radio). Close Companion, Re-open simulator. 

Simulator is meant to be used *in conjunction with* Companion.

snowuyl, 2025-05-02:
Thanks for your reply! Does EdgeTX project provide user manual about how to create radio profiles using EdgeTX companion?

pagrey, 2025-05-02:
The [Official EdgeTX Discussion Thread](https://www.rcgroups.com/forums/showthread.php?3916381-Official-EdgeTX-Discussion-Thread) on rcgroups.com is a great place to ask general questions.

snowuyl, 2025-05-02:
Thanks for your reply!

closed 2025-05-02 by snowuyl