EdgeTX/edgetx #6191: Reassigning axis trimmers to the Radiomaster Pocket

AGE-075, 2025-05-03:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

My RM-Pocket has two excellent trimmer modules based on 5-position push-button joysticks. They support left, right, up, down, and center presses (however, for some reason, the center press is not used in the firmware to center the trimmer). But the problem is that these trimmers are not often used by models with a flight controller. In addition, it users to me that an extra 6-10 controls would be more useful to users than trimmers (if there are currently only 10 controls on the RM-Pocket). Perhaps this will even make the RM-Pocket even better compared to other device options...

### Describe the solution you'd like

I suggest adding a setting that switches the trimmers to the push-button switch mode (With the TRM1/2 x L/R/U/D and TRM1/2 x C buttons), so that it would be possible to add more controls to the RM-Pocket and expand the functionality, as is done on more advanced remote controls. Also, it would be nice to check if there is a central position in the 5-position joystick and also make it functional.

gagarinlg, 2025-05-03:
1. The trimmers are always always used, except for drones
2. the center button is not connected to the MCU
3. for the other trims, this is already possible, when you disable the trims in the flight mode

closed 2025-05-03 by gagarinlg

pfeerick, 2025-05-03:
> 1. The trimmers are always always used, except for drones

And not even for planes with some flight controllers, as they are supposed to be trimmed via other channels, or in some cases, auto-trimmed. 

> But the problem is that these trimmers are not often used by models with a flight controller. In addition, it users to me that an extra 6-10 controls would be more useful to users than trimmers (if there are currently only 10 controls on the RM-Pocket).

As Malte said, you can disable the trims from being used as trims via the flight modes (or configure them there to be in 3POS mode which makes them act like a self-centering 3POS switch). Then you can use them as a switch or other input, in conjunction with logical switches and GV, etc.