EdgeTX/edgetx #6199: LUA scripts not running on V2.11.0

JDessert, 2025-05-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I use LUA script provided by MSH Electronics to configure my Brain2 Flybarless control unit.  Under EdgeTX V2.10.6, the LUA script ran flawlessly.  After upgrading to EdgeTX V2.11.0, script would no longer load.  If I reinstall V2.10.6, script will again load without any issues.  Upgrade to V2.11.0 again, and script won't load.  Code for LUA script I am using can be found here - https://updatebrain.mshbrain.com/software/Brain2OpenTX-EdgeTX.zip.  Specific file that I use within that archive is X:\Brain2OpenTX-EdgeTX.zip\Brain2\HORUS_TX16S_X9D-2019\Brain2_Smart.lua

### Expected Behavior

Expect the LUA script to load and be functional when used on EdgeTX v2.11.0

### Steps To Reproduce

Install Brain2_Smart Lua script scripts\tools directory on TX16S running EdgeTX firmware v2.10.6
Run LUA script from System menu
LUA script loads
Upgrade EdgeTX to firmware V2.11.0
Run LUA script from System Menu 
LUA script doesn't load
Revert to Edge TX V2.10.6 firmware
Run LUA script from System Menu
LUA script loads properly

### Version

2.11.0

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2025-05-04:
2.11 introduces a new LUA version, your supplier needs to update the script for it

55Laurent, 2025-05-04:
> 2.11 introduces a new LUA version, your supplier needs to update the script for it

is there any interest to delete the luac file, to force a new compilation.
I rememeber having such issue with one of the 2.11 RC. the new luac solved it

3djc, 2025-05-04:
It should recompile automatically, but sure, it could help, unless it is a compatibility issue

frankiearzu, 2025-05-05:
I was looking at the zip file...   they are providing the ".luac" as ".lua".. so the ".lua" is not the source code.
They would need to adapt the source code and re-compile it with 2.11 to work.

Also, I notice that they adapted my DSM forward prog and adapted it for their use with spectrum RXs.. They should have provided the source code according to the license.

gagarinlg, 2025-05-06:
> Also, I notice that they adapted my DSM forward prog for their own use with spectrum RXs.. They should have provided the source code according to the license or asked for permission.

If it is GPL licensed, it is enough when they provide the sources to users on request.
Maybe you should just ask them for the source code, based on the license of the work they used to make their own script. Let's see how they react.
When they behave correctly, well everybody can make an error. If they do not behave correctly, nobody should use their products and this should be made more public than a GitHub issue.

3djc, 2025-05-06:
We already had the same issue with Otx with them ..

frankiearzu, 2025-06-12:
I have fixed a few scripts to work on 2.11 with backward compatibility with 2.10.
I think you based your  forward prog code for your "brain" controller on my DSM code.

Contact me directly and I could probably help.

Most of the changes are related to using math (* and /) for bit shift operations.

m-rik-prog, 2025-06-12:
> This is no bug. EdgeTX2.11 uses a new version of lua (5.3) which luac files are incompatible with previous versions.

Of course, they are free to do so. Making software that is not backward compatible, even if it is risky, is still a free choice.
Personally, however, I am equally free to think that it is a bad choice.
In fact, product suppliers like us who use various LUA applications for their products can choose not to distribute and no longer have to continue updating three different packages of their apps, tripling the work involved in releasing updates:
An application package for the new version of EdgeTX,
An application package for older versions of EdgeTX
An application package for OpenTX.
Instead of releasing seven different updates each time, you are forced to release twenty-one updates each time. It is better to abandon the current versions.

pfeerick, 2025-06-13:
If you *choose* to distribute your applications as compiled .luac files
rather than as self-compilable .lua files, yes, the burden on maintaining
multiple binary packages is on you.

Breaking backwards compatibility, especially on embedded systems, is not
really a choice, but an inevitably, if the software is actually being
maintained/actively worked on.

EdgeTX 3.0 will break things even harder in Lua land, as the entire Lua API
is scheduled to be overhauled and consolidated... so the minor issues here
are just a taste of what is yet to come.

gagarinlg, 2025-06-13:
To make it as clear as possible:
By using the source code of the existing script you accepted the license agreement. That license agreement requires you to share the modified source code of your script with all users, that request the source code.

By not doing this, the original author of the script can sue you.

You can now choose to play by the rules and you will get help, or you continue to use the original source code illegally and you won't get any help.

closed 2025-06-13 by gagarinlg