EdgeTX/edgetx #6200: 2.11 NV-14 Non-AETR channels stop changing values

RustyWraith, 2025-05-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

NV-14
All channels other than AETR stop changing values in monitor.
(Not confirmed) Touch sensor stops working.

[model4.yml.txt](https://github.com/user-attachments/files/20029851/model4.yml.txt)

### Expected Behavior

Channels keep registering inputs.

### Steps To Reproduce

Seems random.

### Version

2.11.0

### Transmitter

Flysky NV14

richardclli, 2025-05-08:
> UPD: More precisely, all physical inputs except sticks stop reading values.

Did you check Radio Settings -> Hardware -> Debug (Analog and Keys) to see if hardware has response or not?

RustyWraith, 2025-05-08:
Yes, I did. Only sticks have response until reboot.

richardclli, 2025-05-12:
Is the symptom the same as #6112?

RustyWraith, 2025-05-12:
Exactly. I had this problem with RC2 and RC3 too.

closed 2025-05-13 by richardclli