EdgeTX/edgetx #6203: V2.11.0 crashes RM Boxer

RK-SlowFlyer, 2025-05-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have just put the release version of 2.11.0 onto both my RM Boxer radios (ELRS and CC2500) and both hung on reboot, I had to pull the battery to get back into DFU on both and reload 2.10.6, they are both working now. I did the update to 2.11.0 on my GX12 yesterday and that worked fine.

### Expected Behavior

Reboot and access the new firmware.

### Steps To Reproduce

Exit DFU mode and reboot

### Version

2.11.0

### Transmitter

RadioMaster Boxer

### Operating System (OS)

Windows

### OS Version

Win10

frankiearzu, 2025-05-05:
Cannot reproduce it.. have a boxter internal ELRS, external Multi 4-in-1.
2.11.0 (official) works well

RK-SlowFlyer, 2025-05-06:
Hi Frankie, I hadn't updated the SDCard but I have now and I'm getting the same result. I do the update the same way as you with the bootloader method (trims in + power on, firmware from SDCard/FIRMWARE). The writing finishes and then I exit, I just get the 6P_1 light on and nothing else works, I just have to pull the battery and go back to DFU to rescue the radio.

closed 2025-05-06 by RK-SlowFlyer

x4code, 2025-05-08:
Im having same issue with 2.11 and Boxer ELRS using trims Update(Bootloader) any one got it working?

RK-SlowFlyer, 2025-05-08:
Hello, I couldn't get it to work using the Bootloader method. eventually I tried the DFU mode (I had this term wrong to start with) where you connect the TX with it turned off. I did a cloud build and update via USB. After some trouble with drivers etc. this eventually worked and both my Boxers are now working. I have not had to resort to doing it via this method on any RM TX for years now.