EdgeTX/edgetx #6214: Do you have a plan to port edgetx to linux.

gghyoo, 2025-05-07:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

There are many compact yet high-performance ARM-based SBCs (Single-Board Computers) available on the market. By porting EdgeTX to these devices (such as a ground station system with built-in OPENIPC), it becomes possible to integrate both an HD video receiver (HD VRX) and RC controller functionality within a single ARM SoC.​​

### Describe the solution you'd like

For joystick and button inputs, serial ports or GPIO can be used, while the wireless RF section can directly leverage commercially available transmitter modules (also communicating via serial interface).

raphaelcoeffic, 2025-05-07:
It is *kind-of-the-plan* 😅

The basic idea is to split the code into real-time and UI parts. The real-time would still need to run on an MCU, which would be connected to the UI part via UART / USB / whatever. The UI part could run either on a different MCU (running under FreeRTOS) or Linux. It is not a big issue as LVGL has already some ways to run under Linux (probably based on SDL), which we use to run the simulation.

zandorsp, 2025-05-18:
Still dreaming in the future EdgeTX working something like Klipper on 3D printer: MCU to real-time tasks and UI, management, etc working on an Raspberry or Android cellphone.