EdgeTX/edgetx #6230: Headphone jack

dbock300, 2025-05-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After flashing 2.11.0 the headphone jack has no audio output.

### Expected Behavior

Should work as 2.10.0

### Steps To Reproduce

Plug in headset to jack check for audio.

### Version

2.11.0

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2025-05-09:
It is purely hardware, if you have sound on speaker, you should have on the jack. Did nyou use the trainer jack by mistake ?

dbock300, 2025-05-09:
No if switch back to 2.10.0 if works.

3djc, 2025-05-09:
As I said, there is no software involved, it's purely hardware

dbock300, 2025-05-09:
It's not a hardware issue if it works with 2.10.0 and NOT with 2.11.0. I make no changes to the hardware.

inventor7777, 2025-05-09:
@dbock300 do you have any kind of user volume control enabled? Or have you checked the sound settings on the radio? Make sure nothing got reconfigured.

dbock300, 2025-05-10:
I noticed today that with 2.11.0 my back speaker along with my headphone jack aren't working on my T16S MKII.

However the rear speaker on my buddies TX16S original with 2.11.0. It does not have a headphone jack.

I have been through all the settings and can't find that anything has changed.

3djc, 2025-05-10:
Ok, so it's start to make more sense, it's not headphones that don't work, but you just have no sound.

Could you share your MODEL and RADIO folders ?

dbock300, 2025-05-10:
I have sound but only from the small front speaker.

3djc, 2025-05-10:
Again, all that is purely hardware. There is a single audio output that is hardware routed to the two speakers and headphone jack, nothing Etx does, or can do

dbock300, 2025-05-10:
Sorry to say but as a hardware engineer for 30 years i respectfully disagree.
If the only thing you change is the firmware and make no changes to any hardware configuration then the issue has to be firmware.

Have to tried it on TX16S MKII ?

dbock300, 2025-05-10:
Then you say it's a hardware issue then please explain why it works with 2.8, 2.9 and 2.10 but not with 2.11 ?

raphaelcoeffic, 2025-05-10:
> Then you say it's a hardware issue then please explain why it works with 2.8, 2.9 and 2.10 but not with 2.11 ?

Did you try to downgrade and it works again? I must admit I’m quite puzzled as well how this is supposed to happen.

inventor7777, 2025-05-10:
This is all very weird.

@dbock300 i updated my TX16s MKII to lathe latest 2.11 and the audio works perfectly fine both through the internal speakers and through the headphone jack.

My volume control is set to S2 pot.

dbock300, 2025-05-10:
Then I'm at a loss. My volume control is also on S2.

dbock300, 2025-05-10:
Sorry mine on S1 I'll try changing it.

dbock300, 2025-05-10:
Inventor7777 would you mind contacting me directly via email @ ***@***.*** maybe we can figure this out.

inventor7777, 2025-05-10:
I prefer to keep it on GitHub, hope you’ll understand 😊. Also, I’m not one of the developers, I just happened to see this issue and decided to test on my unit.

inventor7777, 2025-05-10:
If it helps at all, here is my radio configuration. 

[image]
[image]
[image]

dbock300, 2025-05-10:
Did you use tx16s-8369a2e.bin firmware ?

IMG_9571.jpeg (view on web)<https://github.com/user-attachments/assets/de8686d9-897e-4d2c-b99d-5b935974106f>
IMG_9572.jpeg (view on web)<https://github.com/user-attachments/assets/87d36d70-7f4a-4aa2-aa43-46abdabfc466>
IMG_9573.jpeg (view on web)<https://github.com/user-attachments/assets/8fec4c69-cf88-445f-9380-0d937befb92e>

dbock300, 2025-05-10:
My setting are all the same.

inventor7777, 2025-05-10:
Yes, that’s the exact firmware I used. By any chance is yours an older TX16 (MK1)? Maybe there’s a hardware problem that’s reacting with the software incorrectly, or maybe vice versa,

dbock300, 2025-05-10:
No it's a MkII my buddy as an original there is no headphone jack on it.

inventor7777, 2025-05-10:
Got it. Are you using EdgeTX Buddy to flash the firmware? I highly doubt that that’s related in any way but idk.

What does the volume widget show? (The one in the top bar)

[image]

dbock300, 2025-05-10:
No I used bootloader to flash and the widget shows correctly.

IMG_9574.jpeg (view on web)<https://github.com/user-attachments/assets/ffcc1cc8-18d8-4a6b-a2dd-862a3ac09ca6>

gagarinlg, 2025-05-11:
There is a difference between flashing with buddy and flashing with the bootloader.
When you flash with the bootloader, the bootloader is not updated.
Could you please either update the bootloader from within the firmware or use buddy?
Maybe some initialisation was moved to the bootloader.

raphaelcoeffic, 2025-05-11:
> When you flash with the bootloader, the bootloader is not updated.

Maybe the bootloader is even older than we think, if this is the user's preferred update method.

3djc, 2025-05-11:
Front speaker is similar between MK1 and MK2, MK2 feature a back speaker and a headphone jack on the back. Those are hard wired in parallel using the connector shown in this picture. if front speaker is running, but back speaker and headphone aren't then is means something wrong with this wire. There is not control Etx does have over this at all

[image]

dbock300, 2025-05-11:
3djc then please explain why it works with 2.10 and not 2.11.

IMG_0457.jpg (view on web)<https://github.com/user-attachments/assets/2305b379-0f89-4814-be82-f9a39f2ee266>

3djc, 2025-05-11:
I'm telling you what the hardware is, I'm unsure what you do exactly on your side, and please note that no one else has been able to reproduce, which kinda exclude a general software issue

inventor7777, 2025-05-11:
@dbock300 I’d suggest using Buddy and flash the firmware and see if it works after that. If not, there’s got to be something else to do with the volume control method, but idk how

dbock300, 2025-05-11:
Solved in 2.10.2 there was a fix to correct aux port power. Aux one is where I pull 5v for my amplifier connected to headphone jack.

pfeerick, 2025-05-11:
So this is resolved now? Or is there something still outstanding? As is seems it was more than just stock hardware involved here, which is why it could not be reproduced.

dbock300, 2025-05-12:
yes

closed 2025-05-12 by pfeerick