EdgeTX/edgetx #6239: PPM units - global variables too high by a factor of 10

FS4000, 2025-05-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the transmitters basic settings there is the possibility to set the PPM- units
to **0.--** or **0.0** or **us**.
I have tested all three opportunities.
In the **Global Variables** settings there is the possibility to set the units
to **-** or **%** and the precision to **0.--** or **0.0**
I have tried various combinations in Global Variables but the effect is always the same:
**The values seems to high by factor ten.**

### Expected Behavior

The same behavior as before in version 2.10.6

### Steps To Reproduce

In the transmitters basic settings **set the PPM- units e.g. to 0.0**.
Make a test- model and **set a GV to 5.0**.
Put this GV to elevator and set the unit to **-** and the precision to **0.0**.
Mix it with the throttle stick and **the result will be 100% elevator deflection**.

The same behavior can also be seen in Companion.

Best regards and many thanks for your work, Uwe

### Version

2.11.0

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1

philmoz, 2025-05-11:
Can you provide a sample model.yml file that demonstrates the problem please.
I am not understanding your steps to reproduce.

FS4000, 2025-05-12:
Hello philmoz, thanks for your reply.
Here is a testfile. Please look what happens when the throttle is
moved from -100 to +100.
The mixer sets the elevator to 100 although
is only set to a GV- value of 5
If you set the mixer to a value of 5 (not using GV)
the value is as expected (10)

[Test_QX7_Etx_211.etx.txt](https://github.com/user-attachments/files/20159689/Test_QX7_Etx_211.etx.txt)

PR #6250 (2025-05-13): fix(radio): value calculation when GV used as source in weight/offset changed from 2.10
After the change to allow any source for weight/offset, the GV value calculation logic did not quite mimic the 2.10 behaviour.

Fixes: #6239
closed 2025-05-13 by pfeerick