EdgeTX/edgetx #6241: Jumper T20 V2 stops workiing after update to EdgeTX v2.11

ma3yta, 2025-05-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Jumper T20 doesn't work

### Expected Behavior

Jumper T20 works

### Steps To Reproduce

1. Update EdgeTX firmware to v2.11
2. try to lanch controller
3. it doesnt work

### Version

2.11.0

### Transmitter

Jumper T20 V2

### Operating System (OS)

Windows

3djc, 2025-05-11:
It should work, but flashing can sometimes fail. Try DFU flashing

ma3yta, 2025-05-11:
@3djc I used Bootloader method to flash my radio https://manual.edgetx.org/installing-and-updating-edgetx#bootloader-method . It doesn't work. The process finished but after that the radio doesn't start.

pfeerick, 2025-05-11:
Double check if you downloaded t20 or t20v2 firmware onto the handset (and what firmware variant it is currently running).

3djc, 2025-05-12:
> [@3djc](https://github.com/3djc) I used Bootloader method to flash my radio https://manual.edgetx.org/installing-and-updating-edgetx#bootloader-method . It doesn't work. The process finished but after that the radio doesn't start. I rolled back to v2.10.6.

After confirming you got the right version as Pfeerick suggested, please try DFU flashing with radio off. T20 needs to be put in DFU mode using instructions here: https://manual.edgetx.org/edgetx-how-to/access-dfu-and-bootloader-mode

ma3yta, 2025-05-12:
First flashing the bootloader (SD Card -> Flash Bootloader) then update the firmware resolved the problem.

closed 2025-05-12 by ma3yta