EdgeTX/edgetx #6246: Lua script problem after update to 2.11.0

meyer1111, 2025-05-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Some functions are no longer working.
Looks to me like a lua script problem.
For example: app_name = "Flights2" Missing.

### Expected Behavior

No problems from 2.10.6 to 2.11.0 on Display (TX16S MK2).

### Steps To Reproduce

Update from 2.10.6 to 2.11.0.

[image]

[image]

### Version

2.11.0

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

Previously: 2.10.6-lang-custom (14adf047)
language: DE ... bluetooth ...
Custom prebuilt firmware.

After: firmware via Chrome based browser upload.
v2.11.0"Jolly Mon"
Flash firmware via Chrome based browser was not nuccessful?
No more bluetooth?

[image]

inventor7777, 2025-05-14:
I think to get Bluetooth you have to build the firmware yourself AFAIK.

pfeerick, 2025-05-14:
Sort of. You need to use the CloudBuild tab on biddy, and add/configure the
Bluetooth and language flags if you want those options.

I'm not sure how "Flash firmware via Chrome based browser was not
nuccessful?" since showing a screenshot of 2.11 firmware.

For your "noBMP" problem, the Bitmap function had a change in
capitalisation, so your lua script is probably using the wrong form, which
was depreciated a few versions ago.