EdgeTX/edgetx #6249: WiFi Trainer not working on RM Pocket aux port

1nformatica, 2025-05-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am trying to setup a wireless trainer using the "aux 1" port on the Radiomaster POCKET radio motherboard, but it is not working. I have connected an SBUS receiver to it with the TX of the receiver to Rx on the port. The receiver is set for "Inverted SBUS". In the "hardware" tab I have set Serial port AUX1 to "SBUS Trainer". In the "model" I have set Trainer mode to Master/SBUS. I have set a "Special Function" to switch SA so that when down it has the function "trainer" and "Chans", it is enabled. The slave radio is connected to the receiver and I see the Inverted SBUS signal using an oscilloscope on the aux port Rx pin. Another test I made was to assign a mixer to a spare channel and have the source as "TR1" this should show an output when the slave channel 1 (aileron) is moved but it does not. I can connect the same receiver to a Radiomaster TX16S MII and it works as expected.

### Expected Behavior

Connecting an Inverted SBUS receiver to the aux port on the Radiomaster Pocket should work the same as the TX16S MII for the trainer function.

### Steps To Reproduce

1. Connect a receiver with inverted SBUS to the aux port on the RM Pocket Tx on receiver to Rx on the aux port.
2.  In the "hardware" tab set Serial port AUX1 to "SBUS Trainer".
3.  In the "model" set Trainer mode to Master/SBUS.
4. Set a "Special Function" to switch SA so that when latched down it has the function "trainer" and "Chans", it is enabled. 
5. Connect a slave radio to the receiver on the RM Pocket.

### Version

2.10.6

### Transmitter

RadioMaster Pocket

### Anything else?

I have opened a ticket with Radiomaster and here is their response.
"Perhaps the EdgeTX open source project has not fully developed the Uart port function of Pocket, as it is an internal port without an open external interface. We are a hardware manufacturer, and the firmware and all software functions are provided by the EdgeTX open source project system. Therefore, we are not sure if this port corresponds to AUX1 in the menu. You can try to update the Pocket firmware to the latest version 2.11.0 to see if this function can be used normally. Meanwhile, I will also submit the requirements to the production and testing departments for testing. If it still fails to connect to the reverse SBUS, we need to submit this requirement to EdgeTX and look forward to adding and improving this function."

3djc, 2025-05-12:
Master/SBUS is for external module Sbus entry, so not what you want.
You first need to set AUX1 to Sbus trainer, then you should have Master/Serial listed as an option in model setup trainer

1nformatica, 2025-05-12:
> Master/SBUS is for external module Sbus entry, so not what you want. You first need to set AUX1 to Sbus trainer, then you should have Master/Serial listed as an option in model setup trainer.
> 
> Also, I'm not too sure where Wifi comes in play in your issue description

Thanks I now have it working. For WiFi read wireless!!