EdgeTX/edgetx #6257: Audio related feature requests

MrBlackmidi, 2025-05-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Exhibit A: allow the users to increase the gain (in db) for .wav files (for sound packs that aren't full volume)

Exhibit B: allow the user to choose what type of wave form the beep produces (sine, square, sawtooth, triangle and pulse)

### Describe the solution you'd like

Exhibit A: a universal setting (or for sound packs and background seperated) to increase the gain of the audio 
for e.g. sound packs that aren't full volume (which would include the risk of clipping audio (a warning too))

Exhibit B: a setting to change the waveform type the beep plays (mentioned in the top box)

### Describe alternatives you've considered

Only for Exhibit A, increase the gain via scripts, but there aren't currently any downloadable online (afaik)

pagrey, 2025-05-14:
Is it really difficult to find a script to normalize sound files, it even looks like there's one in this project? I don't know why it's there but [maxvolume.cmd](https://github.com/EdgeTX/edgetx/blob/main/radio/util/maxvolume.cmd) looks like it normalizes wav files. There is a wav volume control but correcting the sound files seems like a better thing to do.

MrBlackmidi, 2025-05-18:
i would also like to add for to allow a pitch change when the jog wheel is pressed

pfeerick, 2025-05-18:
> Exhibit A: allow the users to increase the gain (in db) for .wav files (for sound packs that aren't full volume)

As @pagrey said, this is trivial to find online, as well as some examples of how to do it are still present via old scripts in this project still (and also related more to sdcard content than firmware). For example, a quick google search brought up these two results on how to use Audacity (a freely downloadable audio file editor) to both normalise audio files, and how to do batch normalisation. 

https://manual.audacityteam.org/man/normalize.html
https://brianli.com/how-to-batch-normalize-audio-with-audacity/

> Exhibit B: allow the user to choose what type of wave form the beep produces (sine, square, sawtooth, triangle and pulse)

Unless there is really a great need for this, would it not be better/simpler to simply create some wave files with those particular beep sounds/types?

> i would also like to add for to allow a pitch change when the jog wheel is pressed

I'm not sure exactly what you are asking for here? pitch change of what? when? why?

MrBlackmidi, 2025-05-18:
> Unless there is really a great need for this, would it not be better/simpler to simply create some wave files with those particular beep sounds/types? And use the play track SF?

What I mean is that you are able to change the waveform type for scrolling, selecting things, etc

> I'm not sure exactly what you are asking for here? pitch change of what? when? why?

What I mean is that when you are scrolling through the menu(s), it plays at a high pitch for example, when you select things (by pressing the jog wheel for example) it plays at a lower (or higher) pitch
https://youtu.be/bKUNAplCQC0?si=vZi4bziVulfaDn_A&t=31 context

Also for the beep pitch, I would love for it to to have a range of + and -300hz instead of just +300hz

MrBlackmidi, 2025-05-20:
what do you think?

MrBlackmidi, 2025-05-22:
also would also love for an option to allow it to beep when the sys, pg left, pg right, rtn, tele and mdl

MrBlackmidi, 2025-05-22:
It only works for scrolling and pressing on the jog wheel, not on the sys etc buttons (at least in companion, as I don't have a physical edgetx radio)

pfeerick, 2025-05-22:
It works on actual hardware.

MrBlackmidi, 2025-05-22:
oh ok, I realise that I am talking in regards to the B/W screens rather than the colour screens

MrBlackmidi, 2025-05-23:
do you have a say on this? sorry if i have been nagging you

MrBlackmidi, 2025-05-27:
https://github.com/user-attachments/assets/f33a454c-0e4e-4111-8524-182a30294b60 (Video Edited) Demonstration for this feature request

MrBlackmidi, 2025-06-06:
any devs?

MrBlackmidi, 2025-06-09:
are any devs able to check this out? as this hasn't got any other response for a few weeks now

philmoz, 2025-06-09:
EdgeTX development is a small team of volunteers giving up their own time to do this, and we are all very busy.

Someone may look at this if and when they have time; but constantly asking for it is not going to make it happen any faster, and will only serve to annoy everyone.