EdgeTX/edgetx #6261: Bug led charging jumper T15

Mick51, 2025-05-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hello everyone, this message is to warn that on the jumper t15 with v2.11.0 stable the battery charge LEDs no longer light up when the battery is charging.

### Expected Behavior

Hello everyone, this message is to warn that on the jumper t15 with v2.11.0 stable the battery charge LEDs no longer light up when the battery is charging.

### Steps To Reproduce

Charging radio. 

### Version

2.11.0

### Transmitter

Jumper T15

### Operating System (OS)

Windows

gagarinlg, 2025-05-15:
We can only measre the battery voltage and we do not get any additional information about the state of charge, which made the state of charge display via the function switches very inaccurate. This led to complains by users, that the LEDs showed a full battery even thow it was not full. So we decided to remove the inaccurate way of showing the state of charge while charging when the radio is switched off.

closed 2025-05-15 by gagarinlg

pfeerick, 2025-05-15:
In addition, some handsets have a dedicated charge LED, like the T15 (which is why the SYS/TELE keys light up), so the preference is to use those as they are the only reliable indication of charging / not charging.