EdgeTX/edgetx #6263: ST16 sound broken in v2.11.1

richardclli, 2025-05-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Well, the new H7 SPI driver seems having problem in driving the SPI audio in H7 radios.

### Expected Behavior

Fix it!

### Steps To Reproduce

1. Flash
2. Switch on radio
3. The welcome sound .......

### Version

2.11.0

### Transmitter

Flysky ST16

richardclli, 2025-05-16:
@raphaelcoeffic You have ST16, right?  Please fix it.

Boxxxer64, 2025-05-19:
same here. 2.11 work. 2.11.1 sound broken

closed 2025-05-26 by pfeerick