EdgeTX/edgetx #6267: ETX v2.11.00 X12S blank model images

Luznatural, 2025-05-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After upgrading my X12S to ETX v2.11.0 and v2.11.1, all model images are not well displayed, as they show as a white square rather than showing the original picture.
When you sort the models in Manage Models, all pictures are there, but when you select a model, the picture is gone in the home page. 

### Expected Behavior

Model images should be handled correctly

### Steps To Reproduce

1. Starting the radio, the model image has been replaced by a white square.
2. Selecting another model from Model Manager produces the same result
3. In model manager you see perfectly well the images of all your models  

### Version

2.11.0

### Transmitter

FrSky X12

### Operating System (OS)

Windows

### OS Version

11

Luznatural, 2025-05-17:
My firmware is the standard English build

Luznatural, 2025-05-31:
Any solution for this issue?

gismo2004, 2025-05-31:
Have you tried latest nightly? At least for me it seems to be fixed, as I can't reproduce it. Might be this: https://github.com/EdgeTX/edgetx/pull/6296