EdgeTX/edgetx #6268: X12S slow response from start

Luznatural, 2025-05-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After upgrading to ETX v2.11.0 and v2.11.1 the radio responds very slowly to any input, both using touchscreen and buttons. You have to wait a lot to get any response.
Start and shutdown procedures are also slugish, having to wait a lot to see the screen turning on and to see the circle of shut down protocol.
It seems that the radio has ralentized averything.
Flashing v2.10.6 again, the radio recover the normal speed.

### Expected Behavior

The radio should keep working at normal speed.

### Steps To Reproduce

1. Upgrade to 2.11.0 or 2.11.1
2. It takes some seconds for the startup screen to appear
3. Select any menu with touch screen. It takes some seconds for the radio to respond
4. To select any option, very frequently you have to touch the screen several times for getting a response.

### Version

2.11.0

### Transmitter

FrSky X12

### Operating System (OS)

Windows

### OS Version

11

Luznatural, 2025-05-31:
Anyone had the same problem, or it's only me?

jarodgers, 2025-06-07:
I have this issue on the Radiomaster TX16S. Happens on 2.11.0 and 2.11.1. I don't have this issue on 2.10.6.

richardclli, 2025-06-09:
2 known problems that cause this:
1. The misuse of force compilation in LUA widgets, which everytime you start your radio, the lua widgets will be recompile even no change to the source
2. lvgl 8.4 changes

For 1. you can fixed this by yourself, check the BattAnalog and GaugeRotary, in main.lua there are loadScript(...) with flags "tcd", change the flag to "btd" will solve the force recompile problem.

2. Try out the PR https://github.com/EdgeTX/edgetx/pull/6349

jarodgers, 2025-06-09:
@richardclli Awesome, that PR seems to work for me (I am using the build artifacts from philmoz/lvgl-perf, not sure if there was a better way to arrive at the FW file from your link above).

Luznatural, 2025-06-25:
Problem solved with ETX 2.11.2