EdgeTX/edgetx #6269: Add CHOICE option to mixer scripts

wimalopaan, 2025-05-16:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Using inttegers as options is cumbersome. Using meaningfull choice like "Use Code A", "Use Code B", etc. is much more self-explanatory.

### Describe the solution you'd like

Please add a `CHOICE` option for mixer scripts like widgets.

philmoz, 2025-05-16:
Mixer scripts can be used on both B&W and color radios so must be compatible with the features available on both.

B&W radios do not have widgets so do not have the CHOICE widget option type.

3djc, 2025-05-18:
Mixer scripts cannot access lcd or user inputs, or am I missing something  ?

pfeerick, 2025-05-18:
Perhaps you forgot they accept source and integer inputs as part of their configuration? ;)

[image]

3djc, 2025-05-18:
Yes, through the radio UI, not theirs