EdgeTX/edgetx #6287: Channel monitor spacing tweak.

Kevltan, 2025-05-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

@philmoz 
Clipping has now been rectified on the channel monitor screen👍but on a cosmetic note there is a slight offset of the channel bars to the erring to right of the screen and not quite central. Can this be tweaked to the left to centralise things all up as are all other screen displays 😉?

### Expected Behavior

Tweak as above description.

### Steps To Reproduce

Tweak as above description.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

Nightly (e5591839)

philmoz, 2025-05-20:
If you are referring to the space to the left of each channel, that's where the override and inverted icons go.

[image]

Kevltan, 2025-05-20:
Ok Phil, I see it now "doh" silly me! I will close the Ticket