EdgeTX/edgetx #6289: 2.11.0 and 2.11.1 home screen switching second home screen has lot of delay. Pretty laggy

jcer93705, 2025-05-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

2.11.0 and 2.11.1 home screen switching second home screen has lot of delay. Pretty laggy. Some reported on one of the Radio Master group. This is with v2.11.0 and v2.11.1 but v2.10.6 and older version no problem. This is on Tx16s mk2 mck transparent, Tx16 mk1 and Tx16s mk2. So please fix it. Here is a video about the lag. https://youtu.be/YTmQx__8z9Y

### Expected Behavior

Expecting that is instant. 

### Steps To Reproduce

Reproducing by flashing 2.11.xxx version. Going back to older disapear the lag. 

### Version

2.11.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

macOS

### OS Version

15.4.1 (24E263)

### Anything else?

https://youtu.be/YTmQx__8z9Y

qsiguy, 2025-05-21:
I have experienced the same issue.  Specifically when "swiping" the touchscreen to change screens.  If the left/right buttons are pressed the screens will change instantly so seems to be related to the touchscreen in my testing so far.

pfeerick, 2025-05-21:
Touchscreen page switching has swipe animation, which has a higher CPU load
than being able to simply switch the main view from one page to another as
happens when you use the hard keys or special function to switch pages.
2.11.1 should be slightly faster than 2.11.0, but not sure if it can be
improved on any further.