EdgeTX/edgetx #6291: Radio Setup | Backlight | Contrast doesn't load on start, until you look at the setting in menu

Felixrising, 2025-05-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I noticed that the Contrast value doesn't get applied until I go into the menu and scroll down to it, as soon as the cursor goes over the Contrast setting, the actual setting is loaded. Seems like it should be loaded at each TX start.

### Expected Behavior

Load configured constrast value on EdgeTX start.

### Steps To Reproduce

1. modify contrast value
2. shutdown tx
3. start tx
4. note contrast value not loaded
5. go into sys | radio setup | scroll down past "Backlight" to Contrast - 
6. Contrast loads the defined value when cursor reaches the setting.

### Version

2.11.1

### Transmitter

RadioMaster Pocket

3djc, 2025-05-21:
Cannot replicate, working as expected here on newly flashed 2.11.1 on Pocket

Felixrising, 2025-05-21:
Check the video above, from about 30s.

3djc, 2025-05-21:
Try the same on a new blank SD card. It works absolutely fine here, will record a video later

3djc, 2025-05-21:
Ideally BACKUP RADIO and MODELS and delete those in bootloader. Start radio and try.

3djc, 2025-05-21:
https://photos.app.goo.gl/NQ5sU9qE2q1aXMcd9

Felixrising, 2025-05-21:
When I erase my radio.yml, the behaviour is normal. But when I even add the model04.yml, the issue appears again. It's not even loaded, just present. Here are the contents of the model: https://pastebin.com/4iv2nZRw

3djc, 2025-05-21:
I loaded your file, it still works fine here

matejkarasek, 2025-07-17:
Hi, I observe the same behavior on multiple Radiomaster Zorro's, however on my Radiomaster Pocket the Contrast setting works as expected...
The Zorros were flashed with a modified 2.10.5 firmware (custom splash screen added) and matching SD card contents with our models were copied OVER the files on the stock SD card.

Will try the suggested steps when I have time and report back.

PS: 
What I also noticed - after the firmware update, on SOME units, the Contrast value is 128, the screen is black, and I need to manually change it to a reasonable value. Once I go below 30 (and get reasonable contrast values)  I cannot go above 30. 
Might be related to #3542 and #3455, both of which "issues" I also observe