EdgeTX/edgetx #6293: TX16S USB connection causes radio to shutdown

andrzejroszkowski, 2025-05-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

SD Card mode or Joystick causes radio to shut down after connection is selected

### Expected Behavior

USB mode works

### Steps To Reproduce

Turn radio on, connect USB cable, select SD or Joystick mode, confirm selection, for SD card "USB blue screen pops up" then radio "burps" via speaker and dies

### Version

2.11.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### OS Version

Tested with Windows, Linux, even Android phone

andrzejroszkowski, 2025-05-21:
tested as well with "default" sd card contents (created fresh one)

andrzejroszkowski, 2025-05-21:
2.10.6 works fine

3djc, 2025-05-21:
Duplicate #5899