EdgeTX/edgetx #6301: RX numbers show available but are actually in use

LifeLearner4, 2025-05-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I setup a new model and increase the RX number by 1 to find the next available RX number, some RX numbers show as not being used, but in fact are already used.  I know they are already used because 1) I create new RX numbers in order and do not jump around 2) in the EdgeTX Companion I see all my models have RX numbers.  It would be very helpful if, in EdgeTX Companion, a user could click on the heading of model or RX number and have the models be sorted in ascending order or descending order if the heading is clicked again.  Currently, I have to go through all models and count up to find the next available RX number.

### Expected Behavior

I would expect to increase the RX number by 1, over and over again until I get to the next unused and available RX number.  As I increase by 1, I would expect to see which models are using the current number I'm at.  Sometimes I do see which models already use a given RX number but other times it shows the RX number as available and, EdgeTX Companion shows the number is already used.

### Steps To Reproduce

1. Duplicate an existing model
2. Click MDL
3. Choose Internal RF
4. Click on Receiver number
5. Click "+" to increase Receiver number by 1
6. View status of RX number showing "ID is unique" or it will show the model already using the RX number
7. Open EdgeTX Companion
8. Export models from TX
9. Search through list of models to find the true next available RX number to be used

### Version

2.10.6

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

In this example, I copied a previous model that had an RX number of 0.  I clicked "+" to increase the RX number to 1.  The status shows "ID is Unique".  When viewing the models in EdgeTX Companion, I see RX number 1 is already used by the "Hummer" model.  Instead of showing "ID is Unique", it should show that the Hummer model already uses RX number 1.

[image]
[image]

pfeerick, 2025-05-24:
Please keep in mind the same RX number can be used for more than one
protocol, so just because it is "used", does not mean it is a duplicate.

IIRC, double clicking on a column header sorts by that column.

3djc, 2025-05-24:
it is also per module, in addition to being per protocol