EdgeTX/edgetx #6305: Internal GPS not working in EdgeTX 2.11.1

ojoojoo, 2025-05-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I connected a M8Q-5883 M8 GPS on the AUX1 port of my Radiomaster TX16S after I configured the GPS in Ucenter on NMEA output only and the UART1 on 9600-8-N-1.
Under OPTS: I have internalgps.
I enabled the GPS in the transmitter under hardware and started the Internal GPS widget. I don't see any discovery's under Telemetry.
I don't see the GPS communicating with the transmitter, RX and TX are correctly connecter (RX=>TX and TX=>RX)

### Expected Behavior

I expect to read the GPS output on the screen, I get nothing at all.

### Steps To Reproduce

See the text at currect behaviour.

### Version

2.11.1

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2025-05-25:
I'm sorry to report that the issue is in your setup. It is working fine as shown bellow:

[image]

ojoojoo, 2025-05-25:
Oh yes this is really helpfull...

3djc, 2025-05-25:
As a reminder, this is not an end user support tool, just a bug handling tool, and what you report doesn't seem to be a bug...

ojoojoo, 2025-05-25:
So what are your settings used, what type of GPS, did you test this on 2.11.1, because date says 13 apr. what are the uart settings etc. ?
Just saying that it is working doesn't help anybody. It doesn't work in my setup with the settings I posted, and I don't think I did something wrong in my setup so for me it's a bug.

3djc, 2025-05-25:
Tested today, (radio date and time is wrong), 2.11.1 using unblocked m8n gps that was used during this feature development

ojoojoo, 2025-05-25:
Well I can confirm the GPS is working on a Radiomaster TX16S MKII EdgeTX 2.10.1, but not working on a Radiomaster TX16S MKI EdgeTX 2.11.1.
The only two possibilities remain:
A) The UART on the Radiomaster TX16S MKI EdgeTX 2.11.1 is defective (From factory, never used)
B) There is a bug specific for the Radiomaster TX16S MKI EdgeTX 2.11.1

Radiomaster TX16S MKII EdgeTX 2.10.1
[image]

[image]

Radiomaster TX16S MKI EdgeTX 2.11.1
[image]

[image]

3djc, 2025-05-25:
Mk2 and mk1 are electrically 100% identical and therefore share the same firmware

ojoojoo, 2025-05-25:
Than the RX of the UART can be defective, I don't know how to test this quick