EdgeTX/edgetx #6317: Can't edit GV6 in FM1

mha1, 2025-05-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After conversion from 2.10.6 to 2.11.1 GV6 can't be edited in FM1

### Expected Behavior

All GVs can be edited in any flight mode

### Steps To Reproduce

[model15.yml.txt](https://github.com/user-attachments/files/20440662/model15.yml.txt) (delete .txt extension)

https://github.com/user-attachments/assets/390339cf-cec4-4c0e-be8f-ecebeeb5105e

### Version

2.11.1

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2025-05-26:
Isn't GV6 forced by SF7 ?

mha1, 2025-05-26:
Thanks for your response. This is not one of my models. Will check.

mha1, 2025-05-26:
You're right, Thanks for pointing this out and sorry for stealing some of your time.