EdgeTX/edgetx #6319: Gx12 SG & SH switches will not work as 2 pos switch

capposteve, 2025-05-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Setting the SG and SH switches to a 2 position switch instead of toggle does not work as a 2 position switch. All it does is carry on working as a toggle switch. 

Running 2.11.0

### Expected Behavior

Expect the SG and SH swiches to work as a 2 position switch when set to a 2 position switch. 

### Steps To Reproduce

Set switch SG or SH to a 2 position switch and stroll down to debug to check if it works. 

### Version

2.11.0

### Transmitter

RadioMaster GX12

philmoz, 2025-05-26:
SA & SD have LEDs to show their state so they can be safely used as 2POS switches.

You can use a sticky logical switch to create a latching switch from SG or SH.

capposteve, 2025-05-26:
Ok thanks. Not sure how I would go about doing this. Is there a good place to go to learn about this? Thanks

3djc, 2025-05-27:
https://open-txu.org/glossary/sticky/

capposteve, 2025-05-27:
Thanks. I think i have managed it but unsure till i test on the drone later.
i have set v1 to button push in, and V2 to button push in. Looking at the screen it apears to be toggling correctly. However on the main screen, SG still toggles on and off and not like a 2 position swicth. I i have gone into mixes, set channel 11 which was set to the SG button to L01. If im correct this should then work as desired. Fingers crossed.

3djc, 2025-05-27:
No SG remains what it is physically (so still displays as a toggle), and when using it, you can still choose SG that will behave like the toggle it physicaly is, or L01 that emulates a 2POS based on SG

closed 2025-05-27 by 3djc