EdgeTX/edgetx #6330: X-Lite Pro elrsV3.lua Bug if update from 2.11.0 to 2.11.1 (and other lua scripts)

druckgott, 2025-05-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If I use 2.11.0 on my X-Lite Pro all my lua skripts are working. Also the elrsV3.lua from 3.5.5 ELRS Version.
If I use 2.11.1 on my X-Lite Pro all my lua skripts I get not enough memory and sometimes also Syntax error, depending on the lua skript

So lua is not working correctly anymore. It´s on differnt lua scrips.

### Expected Behavior

Scripts rune as 2.11.0

### Steps To Reproduce

Flash 2.11.1 start any lua skript --> error
Flash 2.11.0 start any lua skript --> works

### Version

2.11.1

### Transmitter

FrSky X9 Lite / Lite S, FrSky X-Lite / S / Pro

### Operating System (OS)

Windows

### OS Version

Windows 10

3djc, 2025-05-29:
This is due to https://github.com/EdgeTX/edgetx/pull/6221. Need more time to understand why

druckgott, 2025-05-30:
Ok thx for information

3djc, 2025-05-30:
We are still trying to understand exactly what is going on, but we don't have frsky radios, so it is not easy

3djc, 2025-06-03:
Cool, thanks for the testing!

3djc, 2025-06-07:
@raphaelcoeffic look like your fix fixed one thing but broke another !