EdgeTX/edgetx #6333: Jumper T15 - 2.11.1 - used on two models (F3K) / DLG and had on both models RC outage short after launch.

kijan007, 2025-05-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I flew two models yesterday and today and had similar loss of RC shortly after DLG launch, I was suspecting a bad power connection of first model, but today the same happened on other model. Bench tests have been ok and reliable, but after letting the start button go (TPL) the model became unresponsive and did crash. 

I reverted to 2.10.6 as I never had problems this on this version and it was always reliable. 

Radio : Jumper T15
RX userd: Diversity RX with matek crsf-pwm-10

I attached the model config. I can't explain otherwise than transmitter problem. 

### Expected Behavior

Reliable operation , no freezes of RC controls. 

### Steps To Reproduce

I couldnt't as it randomly happened. I'll not try to reproduce as I don't want to crash another model.

### Version

2.11.1

### Transmitter

Jumper T15

### Anything else?

[ModelsCrashed.zip](https://github.com/user-attachments/files/20535762/ModelsCrashed.zip)

pfeerick, 2025-06-01:
In other words, if you are unwilling to grab a second (micro)SD card, and run that with 2.11.1 (i.e. have 2.10.x config and firmware on one SD card, and have 2.11.x firmware and config on the second, and use the bootloader to switch firmware after swapping the SD cards) and do more tests to find out how to reproduce this, this is unlikely this is going to get resolved with a report like this. Unless, of course, someone else has a similar issue or we are (un)lucky to experience this ourselves. Until then, we need **your** help to work out how to reproduce, diagnose and ultimately, resolve this. 

I honestly do understand that you don't want to crash another model, and aren't suggesting you fly with that setup until you know exactly what the fault is / until it is resolved, but there are other tests that you could still do I'm sure ;)

Johanl1964, 2025-06-11:
You are using a Matek CRSF-PWM adapter, these are unstable if they do not have the latest FW, and can lock up during flight and result in a crash, this is common knowledge, and has also happened to me, but with the latest version FW they are completely stable!