EdgeTX/edgetx #6347: Simulator No Longer Working after Conversion to 2.11

Hartinger, 2025-06-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Upon conversion of an ETX file for the Taranis X9D+ 2019 from 2.10.6 to 2.11.1 the simulator displays the following error.

[image]

When I select Nein, I get.

[image]

and then

[image]

When I open the 2.10.6 version of the file with Companion 2.10, everything is fine.

### Expected Behavior

No error message after invoking the simulator.

### Steps To Reproduce

1. Convert 2.10.6 file using Companion 2.11 (Taranis X9D+ 2019 profile)
2. Try to simulate any model

### Version

2.11.1

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Windows

### Anything else?

Windows 11 Pro Version 24H2

pfeerick, 2025-06-07:
In your first screenshot, it says the ETX file does not match the radio profile selected (X10):

[image]

What does it say next to "Radio" in the Companion titlebar? As if that says FrSky Horus X10 instead of "FrSky Taranis X9D+2019" as shown in my screenshot below, the error is correct, and you need to configure your profile correctly (i.e. `Settings -> Edit Settings...` or the gear cog icon on the toolbar, and select the correct radio type) or switch/create a profile for the X9D+2019 if you have a X10 handset also. I

[image]

Hartinger, 2025-06-07:
I think the profile is configured correctly.

[image]

[image]

The simulator worked fine until 2.10.6, so I guess something went wrong in the conversion.

Hartinger, 2025-06-11:
The problem also occurs when I create a brand new model in companion 2.11.

Hartinger, 2025-06-28:
The problem is still present in 2.11.2.

philmoz, 2025-06-28:
Please upload the affected .etx file so the problem can be investigated.

Hartinger, 2025-06-28:
Here you go (.txt added to the filename to upload).

[EdgeTX_2_11_2_TaranisX9D+.etx.txt](https://github.com/user-attachments/files/20960431/EdgeTX_2_11_2_TaranisX9D%2B.etx.txt)

pfeerick, 2025-06-28:
> Here is an unconverted 2.10.6 file. The simulation works with the 2.10.6
> version of companion but does not work any longer after it has been loaded
> with the 2.11 companion.
>
> EdgeTX_2_10_6_TaranisX9D+_2025-04-30.etx.txt

Since the error message suggests the error is in the profile, rather than
the settings fine, although you have the profile configured correctly, can
you try adding a new radio/profile, and then trying with that? I have had
issues with Companion in the past on Windows where some profile settings
just don't seem right after updates - i.e. wrong SD path used by simulator
(wrong meaning not the path configured in the profile settings).