EdgeTX/edgetx #6362: Linux Appimage Companion Simulator possibly pointing to older EdgeTX version.

Merga189, 2025-06-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

Was going to replace a model image on one model on my radio and in Companion/Simulator.

This new image 

[image]

No problem on the radio.

[image]

The model listing

[image]

and the situation before in Companion/Simulator

[image]

and then selecting the new image

[image]

Strange image in Simulator!

(I tried that on another Debian machine with a fresh installation and no image at all when selecting that image ...)

That are the EdgeTX versions, I am holding on my main machine in case I have to revert.

[image]

I have those two config files in ~/.conf/EdgeTX/

[image]

This is the Companion 2.11.conf

[#8 Companion 2.11.conf.txt](https://github.com/user-attachments/files/20747378/8.Companion.2.11.conf.txt)

To find out, what is happening here, I started Companion 2.11.1 not from the start menu, **but in a terminal and add the end of the terminal protocol the Simulator is pointing not to 2.11.1, but to 2.10.6.**

[#5 terminal.txt](https://github.com/user-attachments/files/20747411/5.terminal.txt)

And that strange image is to be found here in my 'old' data 2.10.6.

[image]

It is not obvious to me, where I made a mistake, which points the Simulator to an old existing EdgeTX version.

Could it be an issue in the Appimage? Running 2.11.1 and the Simulator pointing to 'old' data 2.10.6  ....

br

### Expected Behavior

SIM pointing to data of current version.

### Steps To Reproduce

See current behavior ... 

### Version

2.11.1

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Linux

### OS Version

Debian 12 bookworm

### Anything else?

Took me 2 days to find out ;-)

elecpower, 2025-06-16:
Some of the simulator run settings are saved to the conf file under each profile. You could try a new profile for the same radio.

Merga189, 2025-06-16:
> I checked your configuration and the profile sd path is pointing to 2.11. The simulator run settings are a long binary string and not humanly readable.

I only can see the messages provided by starting Companion 2.11.1 in the Linux terminal.

Merga189, 2025-06-16:
Here the two folders with the 'Companion image' in the foreground and the 'Simulator image' in background

[image]

Quite strange

PR #6415 (2025-07-05): fix(simulator): radio profile sd path
Fixes #6362

Summary of changes:
When starting the simulator use the radio profile sd path instead of the last run saved value.
Note: the profile sd path can be overridden via cli or startup ui.
closed 2025-07-05 by pfeerick