EdgeTX/edgetx #6369: Radiomaster TX16S automatically switches to STM32 BOOTLOADER mode when connected as HID joystick or Mass Storage

nenser, 2025-06-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

**Problem Description:**  
After updating EdgeTX when connecting the radio to a PC in HID joystick mode via USB, after some time (from a few seconds to minutes), the radio disconnects itself, and Windows Device Manager shows "STM32 BOOTLOADER" instead of "HID-compliant game controller."  
This issue prevents using the radio in simulators and other joystick-dependent applications.  

**What I've Tried:**  
- Tested with different USB cables and ports.  
- Updated EdgeTX to the latest version.  
- Changed USB mode settings (Joystick /  Mass Storage).  
- Removed STM32 drivers in Windows.  
- Tried on another PC → same issue.  

### Expected Behavior

that the transmitter will operate in accordance with the mode

### Steps To Reproduce

1. Power on the TX16S.  
2. Connect to PC via USB (HID joystick mode or Mass Storage).  
3. After a while, the radio disconnects and switches to BOOTLOADER mode.  

### Version

2.11.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

windows 10 pro

nenser, 2025-06-17:
RadioMaster TX16S (4 in one) - not MK2

pfeerick, 2025-06-17:
Since this is the TX16MK1, this issue sounds very much like a duplicate of #5899 ... and unfortunately something we have been unable to reproduce. Ttwo users there resolved by replacing the mainboard with a MK2 one. Which from our side, suggests some form of hardware fault/incompatibility, as software-wise, we don't treat the MK2 any differently to the MK1. 

"STM32 BOOTLOADER" indicates the radio has rebooted and is now in DFU mode (since power from the USB was present when the MCU reset).

Did this only start when you upgraded to 2.11 (i.e. same as #5899)? Having said that, there have been isolated reports of something similar as far back as 2.8 (#3039)... another in 2.9 from another user (OP for duplicate issue), then stable in 2.10.x and then pops up again in 2.11.x ... so something strange going on there... 

Someone suggested in #5899 that
> the USB fault disappears if you switch the internal module “OFF”

which if that (I'd also turn off the external module if that is enabled) changes the behaviour for you, would indicate it could be power related (hardware) or a race condition (software).

closed 2025-06-17 by pfeerick