EdgeTX/edgetx #6384: Trainer with SBUS over AUX1 not working since 2.11.0

soumas, 2025-06-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I'm using SBUS signal (generated by a Graupner Hott GR-12 receiver) at AUX1 as trainer "Master/Serial" on my TX16S. After upgrading to the latest Version 2.11.1 I noticed, that my trainer setup no longer works.

Going back version by version (bare firmware and sd card content) I found out, that the latest working version is 2.10.6 with which everything works as usual. A lot has changed in recent versions, but I haven't been able to find any breaking changes that might be related to this. 

Did I miss something, or could it be a bug?

### Expected Behavior

In "Systemsettings --> Trainer" the values should change when the sticks on the student transmitter are moved.

### Steps To Reproduce

1. Clean Installation of EdgeTX 2.11.X on TX12S
2. Systemsettings --> Hardware --> AUX1 = SBUS Input and power on
3. Modelsettings --> Trainer --> Mode = Master/Serial
4. Connect any SBUS generator to AUX1 (power and signal)
--> TX16S does not recognize any student transmitter at all
--> Same words with Version 2.10.6 and lower

### Version

2.11.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Linux

pagrey, 2025-06-22:
When I upgraded I noticed that under model setup, the trainer Ch. Range changed from 1-4 to 1-0 which disabled the trainer output. Selecting 1-4 fixed my issue, it might be worth checking that.

soumas, 2025-06-22:
@pagrey Thank you for the tip. You mean "Model Setup -> Trainer"? Unfortunately, that doesn't solve my problem. I can only change the channel range there if I select the "Slave/Jack" trainer mode. My transmitter, however, is supposed to operate as "Master/Serial."

pagrey, 2025-06-22:
That makes sense, my situation was the slave not generating a signal and you said you have a valid sbus signal input.

soumas, 2025-06-26:
Okay, thanks! Good to know, but even stranger. Since my setup works with older EdgeTX versions, I think it must be a software issue. The only thing I can think of is that my SBUS signal isn't exactly at 3.3 volts, and earlier firmwares were more tolerant in this regard. I'll look into that.

Any other helpful ideas/inputs on how I could analyze the problem?

Thanks, Thomas

mha1, 2025-07-01:
I wonder if and how a GR12 ever worked on AUX1. AFAIK Graupner receivers outputs true (idle low) SBUS. AUX1 requires inverted (idle high) SBUS. @3djc demonstrated this with an ELRS RM ER6 receiver set to inverted SBUS on AUX1.

Graupner receivers work fine in the JR Bay because there true SBUS is accepted while inverted SBUS will not work.

3djc, 2025-07-01:
Indeed ONLY inverted SBUS will work on AUX1, but I (likely wrongly) assumed that this is what the GR12 was giving, since he says it worked in 2.10.

mha1, 2025-07-01:
Only a scope will tell if your inverter works properly.

Here's a great, very clean, and removable option to put have an SBUS capable receiver in the JR bay. It also integrates a 5V regulator which is important for eg ELRS non-PWM receivers: https://radiomasterrc.com/products/radiomaster-sbus-wireless-trainer-modules

pagrey, 2025-07-01:
6579905 seems to have removed the pin pull up. This might be the source of the problem in #6355 as well. Previously init called `
pinInit.Pull = LL_GPIO_PULL_UP;` but I think that was removed. `GPIO_IN_PU` might have been the correct argument in line 42 of aux_ports.cpp instead of `GPIO_IN`.

I don't know if that information was moved somewhere else or if there is a reason not to have it setup the way it was. The inverter circuit used isn't going to work unless the pin has pull up enabled. If you want it to work and are able to do a custom build all you have to do is change the value I mentioned and it should work again.

raphaelcoeffic, 2025-07-01:
> line 42 of aux_ports.cpp

This line initialises the pin that controls the power output in AUX1 / AUX2 (if supported). It is initialised as **output**, not as input, and thus has no pull-up, as output pins are either **push-pull** (default), or **open-drain** (very seldom used).

closed 2025-07-04 by soumas

3djc, 2025-07-04:
No, it simply never worked this way, aux1 is a serial port and cannot hardware decode uninverted SBUS. Never did and never will

soumas, 2025-07-04:
Okay, there's probably no point in explaining again that my setup worked and that I successfully used it for wireless trainer-student communication. Then I must have been a wizard until version 2.11.0 :) Thanks anyway, 3djc!

3djc, 2025-07-04:
Well, I suggest you try it again.. Flashed 2.10.6, connected the same receiver with inverted SBUS, worked fine. Reprogramed it to output SBUS like your GR12, doesn't work...

3djc, 2025-07-04:
Then I guess your inverted was defective, there is simply no other alternative (or like some, you lost power setting on AUX1 connector, but I assumed you would have noticed your receiver was not ON)