EdgeTX/edgetx #6401: CRSF Baro altitude + vario wrong decode

r1000ru, 2025-06-28:
In next lines, Vario part of BaroAltitudeVario frame (0x09) processed as Vario frame (0x07): more when 5 bytes,  bytes after packed altitude, is signed Int16 (in sm/s).
https://github.com/EdgeTX/edgetx/blob/07ce9d77643dfd76b6b897de673ae54903de1983/radio/src/telemetry/crossfire.cpp#L185

But [in the standard](https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#0x09-barometric-altitude--vertical-speed), not two bytes of int16_t are transmitted there, but one byte - the packed speed in int8_t.

Also, according to the standard, packet 0x09  is [implemented in Betaflight](https://github.com/betaflight/betaflight/blob/fd0fd25ef151b74565c07dbf5010b4b4b8ff0acc/src/main/telemetry/crsf.c#L325).

r1000ru, 2025-06-28:
Separately, I want to say that I like the current implementation better. Maybe change the standard?
 - First 2 bytes - packed altitude
 - Next:
   - if 1 byte - packed vario
   - if 2 bytes - vario in uint16_t sm/s

pfeerick, 2025-06-29:
This was **extended** for ELRS in #2253, and any changes to standard need to be done as a PR via https://github.com/tbs-fpv/tbs-crsf-spec, and given this is an ELRS extension, probably discussed with the ELRS devs also. 

Are you saying that if the payload is in accordance with TBS standard, that it *doesn't* treat the third byte of the payload as a packed vertical speed value (which I'm reading the code right, is indeed missing)? This is somewhat of a surprise given that the TBS standard actually references OpenTX in relation to the packed vertical speed processing, but given the fact we don't have access to the standard until recently, not surprising either (as well as the _possibility_ it was updated by TBS along the way...).

r1000ru, 2025-06-29:
crsfPayloadLen get value from CRSF payload.
- For Altitude only is 4 _[FrameType, uint8_t alt_hi, uint8_t alt_low, CRC]_
- For Altitude and Vario as int16_t is 6 _[FrameType, uint8_t alt_hi, uint8_t alt_low, uint8_t vario_hi, uint8_t vario_lo, CRC]_ Vario processed, because **crsfPayloadLen  >5**
- For Altitude and Vario as int8_t is 5 _[FrameType, uint8_t alt_hi, uint8_t alt_low, uint8_t vario_packed, CRC]_. Vario is NOT processed, because crsfPayloadLen  **==** 5

3djc, 2025-06-29:
Yeah, I remember adding various to it, but certainly not packed in one byte. We can add the processing for that one byte value while keeping the apparently unofficial user of 2 bytes, because this seems to have worked for a very long time, so I'm guessing something was (is ?) sending various over 2 bytes

3djc, 2025-06-29:
@r1000ru  could you try a binary out of #6402 ?

PR #6402 (2025-07-01): feat(telem): add support for CRSF 0x09 packed vertical speed
Add support for TBS ALT + VARIO packet frame ([0x09 Barometric Altitude & Vertical Speed](https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#0x09-barometric-altitude--vertical-speed))

This resolves #6401
closed 2025-07-01 by pfeerick