EdgeTX/edgetx #6407: Radiomaster GX12, LED function in "special function" no longer available after upgraded to 2.11.x

cmpang, 2025-06-30:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Radiomaster GX12, LED function in "special function" no longer available after upgraded to 2.11.x

### Expected Behavior

used to able to light up the row of LEDs on SW1-6.
actions in the attached photo no longer work

### Steps To Reproduce

[image]

### Version

2.11.0

### Transmitter

RadioMaster GX12

cmpang, 2025-07-12:
LED flickering is acceptable to me as I am using it to alert an alarm.
Is there a way to revert back to the GX12 stock firmware (feb 2025) so that I can keep this function?

FPVZaphod, 2025-07-20:
See (stalled) discussion here:

https://github.com/EdgeTX/edgetx/issues/6197

Still hope this function will come back :-/

cmpang, 2025-07-21:
> Not stalled, we are working on it, something to control those leds with
> LUA will be part of 3.0, we are still working on the exact details

great to hear that

Mick51, 2025-07-30:
The new TX15 radio has no problem with these LEDs. You could use this as inspiration to put this option back on the GX12. Or will you also remove this function on the TX15 with the next update ?

3djc, 2025-07-30:
Already available on GX12 using a nightly....

Mick51, 2025-07-30:
> * if your switch is set to NONE, then regular "bling" script should be working
> * if your switch is set as a custom switch, new LUA instruction has been added: setCFSLedColor()

I can't get it to work

gagarinlg, 2025-07-30:
Maybe that is because the scripts don't match the FW with the modified LED numbering scheme?

3djc, 2025-07-30:
Have you set your switch to none, because it does work

https://github.com/user-attachments/assets/d8bac4da-10e2-4e49-89b1-4edaf0c15510

Mick51, 2025-07-30:
> Have you set your switch to none, because it does work
> 
>  IMG_0535.MOV

It would be nice to make a video of how to program it, because I've tried a lot of things and nothing works.

Mick51, 2025-07-30:
https://youtube.com/shorts/_SP3r8TvM1c?si=Odr7WBRZpTizPzFz

I tried to do it like in this video but it doesn't work....