EdgeTX/edgetx #6408: latching buttons switchable only on dependence of another switch, but remembering its state

notyourbiz, 2025-06-30:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Radio = Radiomaster GX12, DJI O4pro powered through a RealPit HD (pinio)
Switch to power the O4pro/RealPit = latching button "SD" on GX12 (or any other latching button on any other radio...)

FOV/LOS Flight with/without O4pro

Switching SD by accident during flight

### Describe the solution you'd like

implementing a switch dependency that makes a latching button only _switchable_ if another switch is in a certain state

eg: quad is disarmed (switch 1) - now to select the state of O4pro (latching button) on/off - once armed (switch 1) the latching button remains on/off (weather selected) but is not switchable anymore until quad is disarmed again

 

### Describe alternatives you've considered

i've tried to solve this through logical switches, but it's a pain in the a... to program and never leaded to satisfying results.

even if i solve my "issue" by having an logical switch to deliver my signal (locked once armed), the hardware latching button remains switchable, which means, if i switch it by accident during flight, nothing happens, but as soon as i disarm my logical switch then goes low and my O4pro is unpowered until i switch it on angain.

philmoz, 2025-07-01:
> And if you want to keep using SD, set SD to toggle instead of 2POS in hardware settings, and use sticky if needed.

The problem with this is you lose the visual indication of the state (LED on/off).

3djc, 2025-07-01:
Very true, but I think he is using googles so for him it might not matter at all, and SA or SD are easier to find with goggles on.  Was not meant as 'it is better than..' but more 'here is another option'

notyourbiz, 2025-07-01:
thx guys, both of you, for your effort to support me. i really appreciate that!
in general i agree with both of you, but i do have some objections...

@philmoz: my custom switches are all in use. CS1-3 select different OSD, CS4-6 select different Ratemodes. this is pretty important to me, since thx to elrs i have various quads and fixies on one model. your suggestion works for me for the moment with the exception to switch twice to turn the O4 off if done so during flight by accident... but i'm fine with that for the moment ;-)

@3djc: the visual indication is indeed important to me, since i do los-flights obviously without goggles ;-). but thx anyways, your suggestion is also highly appreciated and will find its way to my setup...

@both: i hope you agree, even if there always is a solution by programming (especially if the state-control for SA & SD is implemented), the whole job for this task could be pretty  challenging for some users (eg like me ;-) ) and a implementation of "my" enhancement could make many peoples live easier, edgetx more powerful and open up numerous new possibilities in progging... ;-)

notyourbiz, 2025-07-28:
could not be happier right now with 0c8c39b. with @philmoz programming sample the SD works exactly like expected. did testing on the nightly and carefully 1st flights and could not find any unexpected behavior/stability issues yet...

thx so much for all yopur work!

s-ol, 2025-09-06:
Great work in #6095, can't wait for 3.0!

> Now create a logical switch to detect when the L03 and SW6 states are not aligned (e.g. L03 is on and SW6 is off, or vice versa). This is only active when the model is armed and will detect pressing the SW6 button when armed.

This explanation sounds very reasonable but I've tried applying it to a slightly different variation (prearm switch needed to arm but not to disarm) and I'm not managing - it seems there's quite a bit of magic in the timing of things here. The way I've been trying is:

- L01 (arm condition): `SW1↓ AND SD↓` (when both switches are down)
- L02 (disarm condition): `|△SW1|>1 AND SD↑` (when arm changes but not prearmed) 
- L03 (arm state): `Stky L01 L02`
- L04 (misalignment): `XOR SW2↓ L04`
- special function: L04: `Push CS SW2`

I've tried different delays for L04 and the special function but I just end up with a lot of seemingly random behaviours and never what I intended, although all the logical switches do exactly what I want.

I wonder if changing "Push Custom Switch" to "Set Custom Switch: ↑/↓/Toggle" would help.
Then the "misalignment" logical switch would not be necessary:

- L01 (arm condition): `SW1↓ AND SD↓` (when both switches are down)
- L02 (disarm condition): `|△SW1|>1 AND SD↑` (when arm changes but not prearmed) 
- L03 (arm state): `Stky L01 L02`
- special function: `!L04: Set CS SW2↑`

Since there is one less logic cycle this seems like it would be less sensitive to delays?

s-ol, 2025-09-06:
Another way to make this a lot more robust would be to just allow setting a different source for the LED lighting information, rather than *reading from and writing to the same switch*, which makes this all very cyclic and astable:

- switch settings:
  - SA (arm): toggle, set LED from L02
  - SD (prearm): toggle
- logical switches:
  - L01 (prearm condition): `SA↓ AND SD↓`
  - L02 (arm state): `Sticky, on=L01, off=SA`
- mixer:
  -  CH5 (arm): L02 

I feel like this would actually be quite an elegant design as well.
I imagine that overriding the LED state in this way should be possible only for switches configured as `toggle`: When a switch is set to `2-position` it has internal state (that must not be hidden from the user). However when the switch is set to `toggle` the LED lighting can be used as a separate output because the user should have no doubt about whether they are currently holding that switch or not 😉.

In fact with this solution the "software 2-position" setting is technically redundant because it is easily implemented using a single logical switch:

- switch setting: toggle, set LED from logical switch
- logical switch: `Sticky SW↓, SW↑`

I do think having this functionality conveniently exposed for the user is valuable, but this could be an opportunity for simplifying the implementation perhaps.