EdgeTX/edgetx #6412: Horus X12S No sounds/beeps after updating 2.11.1 -> 2.11.2

MSmithDev, 2025-07-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating to the new 2.11.2 firmware radio no longer beeps or plays sounds. Flashing back to 2.11.1 returns beeps and sounds.

### Expected Behavior

Normal radio sound functions, key beeps.

### Steps To Reproduce

1. Flash X12S with 2.11.2
2. Copy 2.11.2 SD contents + sound pack
3. Turn radio on 
4. No startup sound or key beeps (with all turned on in settings)

### Version

2.11.2

### Transmitter

Other (Please specify below)

MSmithDev, 2025-07-03:
> Did you flashed both the bootloader and the firmware?

That was the issue thanks!

Luznatural, 2025-07-06:
I have the same problem, and tried several times to flash both the bootloader and the firmware and no success up to now.

MSmithDev, 2025-07-06:
> I have the same problem, and tried several times to flash both the bootloader and the firmware and no success up to now. [@MSmithDev](https://github.com/MSmithDev) would you be so kind to detail how you solved it? Thanks in advance

Its actually still happening for me too. i can sometimes get it to work by reflashing the bootloader and restarting it but it seems like a hit or miss when it works.

Luznatural, 2025-07-06:
You are luckier than me. I never got the sounds back even a single time. I hopr somebody will fix the issue once for all

richardclli, 2025-07-07:
Maybe related to this PR:
https://github.com/EdgeTX/edgetx/pull/6299

But I have tested with my X12S, not sure if it is an intermittent problem

MSmithDev, 2025-07-08:
> Maybe related to this PR: [#6299](https://github.com/EdgeTX/edgetx/pull/6299)
> 
> But I have tested with my X12S before this PR is merged, not sure if it is an intermittent problem

Yeah it seems very intermittent, sometimes when it turn on the radio it works but 3/4 of the time it has no sound at all.

richardclli, 2025-07-08:
One more thing can be checked.  Try to use 2.10.x SD card content and see if there are any differences.  There are some problems in 2.11 SD card content that may prolong the boot up time.  Last time I test my X12S I did not replace the SD card content.

Luznatural, 2025-07-09:
I will try it today

Luznatural, 2025-07-10:
I have tried everything:

richardclli, 2025-07-11:
2.11.1 Is working properly, right?

Luznatural, 2025-07-18:
Any news on this issue?

richardclli, 2025-08-11:
I flashed the X12S using 2.11.2 firmware and 2.11 SD card content, but cannot reproduce the problem.  Please try to flash both the bootloader and the firmware to 2.11.2 and see if the problem gone.

raphaelcoeffic, 2025-08-28:
> This is a real problem not related only with my radio or my configuration, as all 7 people in my club that have the X12S report exactly the same problem.
> We will appreciate if you at least try fixing it, please.

The condition *sine qua non* for a fix is that we can reproduce it with hardware we have. And that is not the case right now. What do you propose?

Luznatural, 2025-08-28:
@raphaelcoeffic, Thanks for responding so quickly. I do appreciate your magnificent work for the benefit of us all.
Sounds are perfectly working on my Radiomater TX16S and my X9D+ with version 2.11.3, but are not working at all in my X12S, so I think is not a generic problem of the versions, but only something related to the firmware for X12S, perhaps affecting at least to some of the batches of the radios.
@MSmithDev mentioned above: "... could possibly be related to this change https://github.com/EdgeTX/edgetx/pull/6299/files ..." I am sure that somebody already checked that. 
@richardclli says above that he couldn't reproduce it.
As the problem is happening on several radios and only in ETX version 2.11, and has not happened before in other versions. Is there a way for me to send you the information on my radio and trying to determine if you can reproduce it in your hardware? I can make a backup using companion both on the radio and in the planes.
As complementary information, I am using my radio in regular En version, and some of the others are using the firmware in Es version, so the problem appears not to be language related.
I am open to any form of collaboration to solve it.
Thanks in advance

3djc, 2025-08-28:
@Luznatural  Could you try this? I have reverted the changes in mentionned PR It cannot be a solution as it breaks another radio, but at least we would know where it comes from

[firmware.bin.zip](https://github.com/user-attachments/files/22021525/firmware.bin.zip)

raphaelcoeffic, 2025-08-28:
@Luznatural the main point here is that the driver for the X12S' sound chip has been rewritten. It works in other radios that use that same chip, and at least on one X12S (Richard's). There could be slight hardware differences inside the same model depending on production batches and/or component tolerances. I quite some suspect some timing issue, but it is very hard to debug / fix without access to that hardware. *"Try and error"* with just wild guesses is gonna take ages and a tremendous effort, considering that there are other radios using that same chip. The main difference between the old and new driver is that the new one has been made based solely on the data sheet from the chip manufacturer, though with higher tolerances on the timings. Currently, it is not clear if this is the culprit or not, or if it is something about the specifics of the X12S(for instance, how the chip is enabled / powered, which is different from other radios).

My current plan is to check with a friend who also has a X12S (one of the first batches; very old radio) if I can reproduce the issue. If I can, then I'll have to borrow that radio and see if I can fix it.

richardclli, 2025-08-29:
It is quite obvious that it is the #6299 PR cause the problem.  I have read the changes and major differences is not to do a soft reset, and skip some unnecessary chip configurations.

Fredflar, 2025-09-02:
Hello, I have an X12S, from a friend, from Open Tx on the Edge TX 2.11.2 Updating.
Now the X12S has no sound. The same problem as @luznatural.
Hope a solution will be found soon.
At the next meeting I will try to reset the X12S to Edge TX 2.11.1.

richardclli, 2025-12-29:
However, I cannot reproduce the problem with my X12S.  It is a bit hard to track down the problem.

corratec100, 2026-01-01:
For Info: I had problems with the sound in versions 2.11.2 and 2.11.3. I have now tested version 2.12.0-RC1. The sound works with this version.

richardclli, 2026-01-05:
Interesting that it is fixed in 2.12.0 ........  Let me see if anything about sound is changed between these 2 branches.

raphaelcoeffic, 2026-03-31:
> I also have a Horus x12s and am experiencing the same sound issues with ETX 2.12.0 as @Lutznatural. For your information: Every time the transmitter was charged and then turned on, there was no sound. If the transmitter was then turned off and on again several times, the sound would suddenly return. This issue was reproducible. Thank you very much for your efforts.

Did you try this: https://github.com/EdgeTX/edgetx/issues/6412#issuecomment-4155071755

raphaelcoeffic, 2026-04-10:
> Comment avoir le 2.12 en fr + le fix-https://github.com/EdgeTX/edgetx/issues/6412

This fix will be in the next 2.12 and 2.11 releases. Then you can download a french firmware using either Companion or Buddy+Cloudbuild.