EdgeTX/edgetx #6417: Include STM32 HAL drivers as submodules

fpenard, 2025-07-03:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

The STM32 HAL drivers sources are directly copied in sub-directories in `radio/src/thirdparty/`.
This makes the evolution of changes complex and does not allow a direct link with the STMicroelectronics github repositories.

### Describe the solution you'd like

This proposed change consist to replace STM32 HAL drivers directories (`radio/src/thirdparty/STM32F2xx_HAL_Driver`, `radio/src/thirdparty/STM32F4xx_HAL_Driver`, `radio/src/thirdparty/STM32H7xx_HAL_Driver` & `radio/src/thirdparty/STM32H7RS_HAL_Driver`), by the next repositories included as submodules:
- https://github.com/STMicroelectronics/stm32f2xx-hal-driver
- https://github.com/STMicroelectronics/stm32f4xx-hal-driver
- https://github.com/STMicroelectronics/stm32h7xx-hal-driver
- https://github.com/STMicroelectronics/stm32h7rsxx-hal-driver

### Additional context

I perform a diff between the STM32 HAL drivers sources and STMicroelectronics github repositories, to evaluate the risk to implement this change.
Only minor changes observed on stm32f4 drivers. No change observed for the other ones.
In my opinion, these no risk to implement this change.

**edgetx release considered**
https://github.com/EdgeTX/edgetx.git
- sha : d825cb9a38579da47d8ecbae0cb08d5322312326
- date : Date:   Wed Jul 2 12:02:26 2025 +1000

**stm32f2xx-hal-driver release considered & diff result**

https://github.com/STMicroelectronics/stm32f2xx-hal-driver
- sha : commit f6f6991251020ea5f41db85a85adbf81f574ca8c
- tag: v1.2.8
- date:   Thu Apr 27 17:51:41 2023 +0100
==> No difference with : radio/src/thirdparty/STM32F2xx_HAL_Driver

**stm32f4xx-hal-driver release considered & diff result**

https://github.com/STMicroelectronics/stm32f4xx-hal-driver
- sha : c91c4a744c5bcf881c5172a4f064bbc5eb7a5e39
- tag: v1.7.13
- date:   Mon Aug 9 12:59:33 2021 +0100
==> 3 minor changes with : radio/src/thirdparty/STM32F4xx_HAL_Driver
==> From my point of view, no functional impact.

In stm32_hal_legacy.h :
```diff
#define FLASH_ERROR_NONE              HAL_FLASH_ERROR_NONE
#define FLASH_ERROR_RD                HAL_FLASH_ERROR_RD
#define FLASH_ERROR_PG                HAL_FLASH_ERROR_PROG
-#define FLASH_ERROR_PGP               HAL_FLASH_ERROR_PGS
+//#define FLASH_ERROR_PGP               HAL_FLASH_ERROR_PGS
#define FLASH_ERROR_WRP               HAL_FLASH_ERROR_WRP
#define FLASH_ERROR_OPTV              HAL_FLASH_ERROR_OPTV
```

In stm32_hal_def.h :

```diff
/* Exported macro ------------------------------------------------------------*/

-#if !defined(UNUSED)
#define UNUSED(X) (void)X      /* To avoid gcc/g++ warnings */
-#endif

#define HAL_MAX_DELAY      0xFFFFFFFFU
```

In stm32_hal_def.h :

```diff
HAL_StatusTypeDef HAL_DAC_Start_DMA(DAC_HandleTypeDef *hdac, uint32_t Channel, uint32_t *pData, uint32_t Length,
                                    uint32_t Alignment)
{
+  HAL_StatusTypeDef status;
-  HAL_StatusTypeDef status = HAL_OK;
  uint32_t tmpreg = 0U;

  /* Check the parameters */
```

**stm32h7xx-hal-driver release considered & diff result**

https://github.com/STMicroelectronics/stm32h7xx-hal-driver
- sha : commit b716379524ba3549e4db11804889776ffeb239d1
- tag: v1.11.1
- date:   Thu Dec 8 09:25:51 2022 +0100
==> No difference with : radio/src/thirdparty/STM32H7xx_HAL_Driver

**stm32h7rsxx-hal-driver release considered & diff result**

https://github.com/STMicroelectronics/stm32h7rsxx-hal-driver
- sha : commit d55e2661c5bb338778c955cb89e57cae2a3746de
- tag: v1.0.0
- date:   Thu Mar 21 14:14:13 2024 +0100
==> No difference with : radio/src/thirdparty/STM32H7RS_HAL_Driver

fpenard, 2025-07-07:
Thank you for your feedback.
I will go further in analyzing the changes, in order to help better assess the impacts.

Adding submodule update to CMAKE [submodule update management itself at build time](https://cliutils.gitlab.io/modern-cmake/chapters/projects/submodule.html), is interesting. I will study it.

pfeerick, 2025-07-09:
> I disagree, I would prefer to fork the repo, make the patches and possibly make a PR against the ST repo

For changes that should go upstream (i.e. bugs in their code), most definetley you want to push for them to be fixed upstream. For changes that are specific to our code (i.e. define options, configuration file changes)... the question is then simply of which is the least _additional_  maintenance/easier to track - a seperate repo or a patchfile? If you think that is a seperate repo, fine by me... I probably won't be the one who'll be maintaining it :-P (well, until I do end up having to 😮 ).

fpenard, 2025-08-17:
> [@fpenard](https://github.com/fpenard) are you active on our discord server? Almost all of our discussions regarding the continued development are happening there.

I'm not connected to edgetx discord sever yet. I will look to join you on discord, to be aware of current developments.

fpenard, 2025-08-17:
Thanks all for your feedbacks.
I agree with @gagarinlg. A fork of ST repo,to apply the patches, will be easier to maintain.
I think it's more consistent to perform this fork in https://github.com/EdgeTX/.
Who would have the access to do that?