EdgeTX/edgetx #6421: Companion for Linux - Appimage problem

AlanT0803, 2025-07-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Since version 2.11 (i.e. 2.11.0, 2.11.1 and 2.11.2), the Linux Appimage for Companion appears to be 'broken'.

The Appimage will not register or run under Fedora 42 or Debian 12. Version 2.10.6 works correctly.

### Expected Behavior

The Appimage should register and run as it did prior to version 2.11.

### Steps To Reproduce

Attempt to register and run the Appimange

### Version

2.11.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Linux

### OS Version

Fedora 42, Debian 12

Miwer, 2025-07-07:
Same on Manjaro with
AppImageLauncher version 2.2.0 (git commit 9ca001e), built on 2025-07-07 16:30:45 UTC

Other appimages work fine - including Companion 2.10.4

```
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: fuse: memory allocation failed
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: squashfuse 0.5.2 (c) 2012 Dave Vasilevsky
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: Usage: squashfuse [options] ARCHIVE MOUNTPOINT
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: (null) options:
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o offset=N            offset N bytes into ARCHIVE to mount
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o subdir=PATH         mount subdirectory PATH of ARCHIVE
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o notify_pipe=PATH    named pipe that will receive 's' (success)
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:                            or 'f' (failure) when the mountpoint is ready
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o timeout=N           idle N seconds for automatic unmount
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o uid=N               set file owner to uid N
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o gid=N               set file group to gid N
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: FUSE options:
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -h   --help            print help
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -V   --version         print version
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -d   -o debug          enable debug output (implies -f)
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -f                     foreground operation
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -s                     disable multi-threaded operation
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o clone_fd            use separate fuse device fd for each thread
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:                            (may improve performance)
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o max_idle_threads    the maximum number of idle worker threads
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:                            allowed (default: -1)
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o max_threads         the maximum number of worker threads
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: fuse: memory allocation failed
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: squashfuse 0.5.2 (c) 2012 Dave Vasilevsky
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: Usage: squashfuse [options] ARCHIVE MOUNTPOINT
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: (null) options:
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o offset=N            offset N bytes into ARCHIVE to mount
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o subdir=PATH         mount subdirectory PATH of ARCHIVE
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o notify_pipe=PATH    named pipe that will receive 's' (success)
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:                            or 'f' (failure) when the mountpoint is ready
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o timeout=N           idle N seconds for automatic unmount
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o uid=N               set file owner to uid N
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o gid=N               set file group to gid N
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: FUSE options:
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: squashfuse 0.5.2 (c) 2012 Dave Vasilevsky
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: Usage: squashfuse [options] ARCHIVE MOUNTPOINT
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: (null) options:
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o offset=N            offset N bytes into ARCHIVE to mount
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o subdir=PATH         mount subdirectory PATH of ARCHIVE
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o notify_pipe=PATH    named pipe that will receive 's' (success)
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:                            or 'f' (failure) when the mountpoint is ready
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]:     -o timeout=N           idle N seconds for automatic unmount
[... cut ...]
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: Cannot mount AppImage, please check your FUSE setup.
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: You might still be able to extract the contents of this AppImage
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: if you run it with the --appimage-extract option.
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: See https://github.com/AppImage/AppImageKit/wiki/FUSE
Jul 07 18:36:08 localhost EdgeTX_Companion_2.11.2-x86_64.AppImage[13883]: for more information
``

elecpower, 2025-07-08:
A quick Internet search indicates it is an issue in a particular version of AppImageLauncher refer https://forum.cursor.com/t/0-47-9-failed-to-launch-fuse-memory-allocation-failed/68394

AlanT0803, 2025-07-08:
Interesting.
I was using AppImageLauncher under Fedora 42 and Companion 2.11.x would not run.
However, I also have a machine running Debian 12 which does not have AppImageLauncher installed. Companion 2.11.x wouldn't run on that machine either, which make me think the problem was with the Companion AppImage.
I have now removed AppImageLauncher and installed GearLever. Companion 2.11.2 now runs under Fedora 42.
I will play around with the Debian install to see if I can figure it out.
Thanks for the help, my apologies for jumping to the wrong conclusion.