EdgeTX/edgetx #6425: Method to mass delete log files

aquaaddict, 2025-07-06:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

A large volume of log files becomes unwieldy and slows down log file viewer etc. log file don't need to be kept forever, but currently the only way to delete them is 1 at a time thru via the UI, or connect to a PC and delete the files from card. 

### Describe the solution you'd like

Please can we have an option to auto delete old log files (e.g. on shutdown), maybe have a configurable time period so only delete log files older than x days are retained. 
It might need to also delete the LogViewer.csv file (if it exists) so it is rebuilt and does not contain references to log files that are since deleted.

### Describe alternatives you've considered

Alternatively it could be a LUA script to delete old log files which could be manually run (or could be triggered by an event if there was a version that didn't display to the screen).

pfeerick, 2025-07-06:
In the interim, you might be interested in this Lua tool

https://github.com/EdgeTX/lua-scripts?tab=readme-ov-file#logmanager

aquaaddict, 2025-07-12:
> In the interim, you might be interested in this Lua tool
> 
> https://github.com/EdgeTX/lua-scripts?tab=readme-ov-file#logmanager
> […](#)

Thanks, that's a great help. IF there was a way to trigger it on shutdown with pre-chosen options it would be perfect