EdgeTX/edgetx #6429: Flashing FW to external 4in1 module not working

TX-Chris, 2025-07-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When trying to load MPM firmware to JP4in1-SE external module, error message "Warning - Firmware update error - Device not responding" appears on screen. 
Configuration: 
Tested on EdgeTX 2.11.0 and 2.11.2, with TX16S-RGB (MKII)
Jumper MPM JP4in1-SE
tested with FWs: AETR-Air-v1.3.4.31, AETR-LBT-v1.3.4.31, AETR-v1.3.3.33

FW flashing is working fine on openTX 2.3.15 on a second TX16S. 

Otherwise, MPM module works fine for both, TX and telemetry. 

### Expected Behavior

Flashing FW to external MPM has to be possible. 

### Steps To Reproduce

1. make sure MPM is installed and operational
2. turn radio on and go to: SYS -> SD CARD -> FIRMWARE
3. select desired FW to be installed on MPM
4. hit "RETURN" and select "Flash External Multi" to start the FW installation
5. screen shows "Flash Device  --  Device Reset". MPM green and red LED go off for approx 3 s, then green LED turns permanently on and red LED starts flashing for another 5 sec, then both go off again, staying off until message "Warning - Firmware update error - Device not responding" on screen is confirmed by user. 

### Version

2.11.2

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2025-07-08:
Just tested TX6SMKII with external MPM, flash is working fine here

TX-Chris, 2025-07-09:
Which MPM are you using?

Tried it over on a second TX16S running edgeTX v2.9.4 .... same result - not working.

kochpet, 2025-07-22:
Looks to me like this is the same problem as #6295  which I reported some time ago.