EdgeTX/edgetx #6434: "USB Joystick" menu disappeared in v2.11.x

peterzdeb, 2025-07-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The "USB Joystick" menu is available on model setup page for firmware versions up to 2.10.6. I can configure my RC as a joystick in "Advanced" mode.

Remote Control: RadioMaster TX12 MK2
SD Card: 4Gb

Tried almost all firmware versions. SD card format and data restore doesn't help.
EdgeTX versionn 2.11.0 and up do not show the "USB Joystick" menu.

### Expected Behavior

When selecting a model and visiting "Setup" page the last menu item is "USB Joystick".

### Steps To Reproduce

1. Flash EdgeTX v2.11.0 on RadioMaster TX12 MK2
2. Select any model 
3. Visit "Setup" page
4. Scroll down to the last menu item - "USB Joystick" must be there as in previous versions.

### Version

2.11.2

### Transmitter

RadioMaster TX16S / TX16SMK2, RadioMaster TX12 / TX12MK2

### Operating System (OS)

macOS

pfeerick, 2025-07-11:
This may not be possible to change - this was deliberately done in #5443 for STM32F4 MCU targets with only 512KB of flash (program) storage. We were running into issues whereby some of these targets would not have enough space or translated firmware couldn't be built, and the advanced usb joystick feature uses a bit of flash up. 

The TX12MK2 is one of these. If you want a 2.11.x firmware only in english... it looks like there is still 3KB (99.26% flash usage) if enabling `USBJ_EX`, so can either provide you with a firmware, or you can build it yourself via GitPod - just need to comment these two lines out.

https://github.com/EdgeTX/edgetx/blob/a7fc3dbc3cc621923d81ae6bc9fc1a000ffeaf28/radio/src/targets/taranis/CMakeLists.txt#L547-L548

peterzdeb, 2025-07-26:
Hi @pfeerick.
Thanks for providing details and confirming that's a known issue.
That's unfortunate to miss such an important functionality just because other not always useful features comsume all the flash. The advanced joystick is a great option to connect joystock on any platform and make it working consistently.

By "firmware only in english" do you mean some other flags must be unset for firmware compilation?
I'll try building the firmware myself

pfeerick, 2025-07-26:
By default the firmware builds in english... if you want any other language
you need to specify the `TRANSLATIONS` parameter, i.e. `-DTRANSLATIONS=DE`
for German. But you may then also find that you won't be able to build for
that language, as the translations for some languages are larger than
English/some others.

johyphenel, 2026-10-05:
It's frustrating to lose the ability to set up trim as buttons without the joystick mode. I want to use L Trim as a dpad for when I use sims on steamos connected to my tv.

jpmche, 2026-10-08:
I just ran into this issue after upgrading the RM Pocket to the latest firmware. It'd be awesome if there were a solution that lets us upgrade to a newer version of EdgeTX without losing the feature.

pfeerick, 2026-10-08:
There are now options to turn off support for RF modules that you don't use/have on cloudbuild, which opens up the option of adding one to turn "ON" the advanced joystick capability... (since you would be able to turn off something else to get the space needed to enable it). I'll look at that to be included in 2.11.9.

jpmche, 2026-10-08:
I built versions 2.11.7 and 2.12.3 with the flags Pxx2=off/Ghost=Off on Cloudbuild, but the USB Joystick option is still missing from the Model Setup page. Should I build a different firmware version? Thanks!

pfeerick, 2026-10-08:
> which opens up the option of adding one to turn "ON" the advanced joystick capability... 

Note I said ADDING the option... ;) You need to be able to turn it on, you can't just turn the other capabilties off and it magically appear  ;)