EdgeTX/edgetx #6442: Telemetry problems when using internal and external module together

WimHWimH, 2025-07-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using the internal (CC2500/LBT/Hott) module and the external module (IRX4MPM/Air/MPX) together on my Radiomaster Pocket , only the external module seems to report telemetry (even though the RX has no telemetry) . After changing (in Companion) the external module to no telemetry and leaving the internal one on telemetry, neither module has telemetry. After the change, I can see the telemetry status in the TX ( before the change only for the external module) , but it cannot be changed (greyed out). Companion and TX 2.11.1, latest MPM module version on both modules.

### Expected Behavior

I expected the TX to report the same I entered in Companion (telemetry only for the internal module) and/or to be able to change the telemetry behaviour inn the TX. Very little can be found about using two MPM modules together.

### Steps To Reproduce

1. activate both internal and external MPM modules
2. Check telemetry status ( off for internal, on for external)
3. Change telemetry to off for the external module  (on for internal one) in Companion
4. copy change to TX
5. Both modules are "no telemetry" now, cannot change in TX

### Version

2.11.1

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

### OS Version

11

### Anything else?

I want to use the external module for my F5B control receiver, but would like telemetry from my internal module still.

WimHWimH, 2025-07-15:
Just checked again,everything works as it should. Did not change anything , tried switching rx and tx on and off a couple times without success , now, a few days later, it works as expected.

pfeerick, 2025-07-19:
Just a FYI, don't confuse module and over the air (OTA) telemetry... one is for the RF module (whether internal or external) to be able to talk back to the handset (i.e. things like the "Module Status: V1.3.4.0 AETR" line, protocols supported, OTA telemetry) and the other "Disable Telemetry" option (which actually needs to be removed as Companion shouldn't be showing that if the handset isn't a FrSky one - per #3637) which was really only for FrSky handsets since the S.Port line for the external module was shared with internal module. When you're running two MPM, I don't think you should be having any major issues, but it has been a while since I've tried that. It is a bit of a mess running two modules simultaneously, as if you need to do sensor discovery, you've got twice the amount of telemetry sensors that could be messed up (i.e. indexes will change for all input/mixer/ls/sf/widgets that use them).

WimHWimH, 2025-07-21:
I now think my "problem" might be related to this :  https://github.com/EdgeTX/edgetx/issues/6400 . Apparently this was fixed? Any idea in what release I can expect the fix? (I'm on 2.11.1) My Radio is thae Radiomaster Pocket though...