EdgeTX/edgetx #6475: Spectrum Analyzer not working with External MPM on X9DP @ETX2.11.2

Kai2510, 2025-07-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The Spectrum Analyzer (EXT) will display like this no matter which protocol selected:
[image]

The script's other functions runs normally except the spectrum display. Also, the MPM runs normally when transmittering signal.

### Expected Behavior

Spectrum analyzer works normally just like what it was on OTX 2.3.11.

### Steps To Reproduce

Go to TOOLS page and open Spectrum (EXT).

### Version

2.11.2

### Transmitter

FrSky X9D+

J-Sorenson, 2025-08-25:
Did the spectrum analyzer work on all 4-in-1 chips, or just the CC2500?  It's been a long time since I bothered to use it.

Kai2510, 2025-08-31:
> Did the spectrum analyzer work on all 4-in-1 chips, or just the CC2500? It's been a long time since I bothered to use it. Last time I was using the internal FrSky ACCST module(CC2500) with the analyzer, so I've never tried it with the multi.

My external MPM is 4in1, running old version firmware (maybe v1.39?).

Kai2510, 2025-11-18:
Update: downgraded to ETX 2.10.6, updated MPM firmware to 1.3.4.31, still doesn't work. After disabling internal RF, it works.

alm-nl, 2025-12-15:
It works on EdgeTX 2.11.4 on my X9DP with iRangeX IRX4+.