EdgeTX/edgetx #6479: change model takes long time since 2.11

ParkerEde, 2025-07-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have carried out a direct comparison of the model change once with 2.10.6 and 2.11.2.
With 2.10.6 it takes 1.2 seconds
with 2.11.2 it takes 4.1 seconds
It has been changed from the same model to the same new model on the same SD card with the same RM TX16S.

Why does it take almost 4 times longer with the new version than with 2.10.6 ? Is there any potential to get the old speed back?

### Expected Behavior

2.11 should be able to load the models just as quickly as 2.10.6 can.

### Steps To Reproduce

Compare the model change speed under the same circumstances from 2.10 to 2.11

### Version

2.11.2

### Transmitter

RadioMaster TX16S / TX16SMK2

philmoz, 2025-07-28:
Do you have multiple main views/screens with the built in widgets (Outputs, Model Info, Timer, Value, etc)?

philmoz, 2025-07-28:
In 2.10 the built in widgets did not use LVGL, like Lua widgets they redrew their content completely on every cycle.
So while they were quick to create, they used a lot of CPU cycles when on screen.

In 2.11 everything in the firmware has been converted to LVGL. This means the widgets are much faster when running as only changed areas need to be redrawn. The drawback is it takes longer to build and layout the widget when it is first created - LVGL can be a bit slow here.

When you change models, the screens are re-created and new widgets created - which can take more time on 2.11.

PR #6480 will address some of this by delaying the build/layout of the widgets until they are needed. The downside is you may see the screen appear before the widget content gets shown.

If your Lua widgets are using LVGL then they may also be slowing down the model switch (for the same reason). I need to see if delay loading can also be applied here.

ParkerEde, 2025-07-29:
Thank you very much for the detailed information. Do you need anything else from me for further review?

philmoz, 2025-07-29:
> Thank you very much for the detailed information. Do you need anything else from me for further review?

If you are able to test the latest 3.0 nightly it would be nice to know how it compares (it contains PR 6480 changes).
Be sure to have a good backup as there are YAML changes in 3.0 that may not be compatible with 2.11.

philmoz, 2025-07-30:
With 5 screens, it's the slider and trim UI elements that is taking extra time. Same reason - the LVGL build/layout time.

By delaying the load of these I can get the model switch down to 1.3 secs on my TX16S, with the tradeoff that the trims and sliders don't appear initially. This also affects switching screens - the first time you change to another screen there is short delay before all the UI elements are created.

philmoz, 2025-07-31:
I was able to speed up the Trim and Slider UI creation without resorting to delay loading.

Could you please test the attached TX16S firmware on your radio.

[TX16Sfirmware.bin.zip](https://github.com/user-attachments/files/21522022/TX16Sfirmware.bin.zip)

ParkerEde, 2025-08-01:
perfect, now it takes 1.7 seconds. many thanks.
Can we have this solution in 2.11.3 ?

closed 2025-08-02 by pfeerick