EdgeTX/edgetx #6491: Internal gps support for gx12

kakashi2018, 2025-07-31:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Internal gps not working with gx12 radio.

### Describe the solution you'd like

Radiomaster gx12 has an open uart inside the radio and would like to add gps module to maintain time with out manually changing it over and over again. This is supported for tx16 radio

3djc, 2025-08-01:
It hasn't been added to BW radio because no one so far has been able to physically install a GPS module in a BW radio, since space is to tight on those, and since those module need a view to the sky, which complicates installation further. I have made a binary for GX12 that should support it, could you test that ?

[firmware.bin.zip](https://github.com/user-attachments/files/21543001/firmware.bin.zip)

kakashi2018, 2025-08-02:
There is just enough space for foxeer m10 non gps module to slide and go front facing inside the radio. I have tried your firmware and it didn’t work. Also tried changing baud rate to 9600 still didn’t work. This gps is both nmea and ublox supported