EdgeTX/edgetx #6514: MT12 Companion is not displaying MPM DSM subptype 2SFC

elekSENDER, 2025-08-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Companion is not displaying Multi DSM subptype 2SFC with a MT12 profile although it is displayed in the MT12

[image]

### Expected Behavior

Companion should display and save subtype 2SFC.

### Steps To Reproduce

Program a model on a MT12 with 4in1-MPM (SFC-firmware) with DSM/subtype 2SFC.
Read it to companion and there is an empty subtype-box, in the dropdown there ist no posibility for 2SFC.

### Version

2.11.2

### Transmitter

RadioMaster MT12

### Operating System (OS)

Windows

### OS Version

Win 10

### Anything else?

Tried the same in a TX16s with the AIR-firmware in the 4in1 MPM. After selecting 2SFC the "Module Status" becomes "Protocol invalid".
For me it seems there is any dependency on the MPM firmware in the transmitters by selecting the subtypes.

3djc, 2025-08-13:
There is no way for companion to know what protocol are (or are not) in a given multi protocol module, and there is no way they can be kept in sync. On the radio, there is a dialog between the radio and the module so the module can tell the radio what protocol are in the module

elekSENDER, 2025-08-13:
OK, understood. 

So where is the source for the items in the dropdown-box for subtype in companion?
This source should be expand with the option 2SFC, companion should only be able to handle (read/write) the same possible parameters as the radio.

This was the primeraly reason for me to open this issue, a sync between companion and MPM is not necessary.

pagrey, 2025-08-13:
Companion is obviously out of date. All the protocols are listed in the DIY project in the file [Multiprotocol.h](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multiprotocol.h) It isn't impossible to keep them in sync.

This project is full of external sources, it's so unfortunate that such a simple request is treated like this. Wouldn't it just be easier to say you don't have time or it isn't a priority instead of suggesting there is no way they can be kept in sync?

pfeerick, 2025-08-14:
It was probably just a lost in translation meaning... i.e. there is no way to keep the specific list of protocols that your currently installed MPM firmware (as the list of protocols/subprotocols is *MPM firmware dependent*...) in sync with Companion. Companion doesn't know - it only knows what the latest list of protocols is (as of when it was compiled. If you run air only firmware, it will allow you to pick both air and surface, same for FCC only or LBT only firmware options. All we can do is update the list periodically so that *all* the options are present in Companion.

3djc, 2025-08-14:
Yes, we have people raising issues because they see protocols in companion then it does not work on the radio. I thought about a way to sync using CLI, but it ends up been to complicated to operate to be helpfull

elekSENDER, 2025-08-14:
Peter, thanks for translating my thoughts.
I  just wanted to make a hint to see a parameter in the radio is not available in companion.

I know this is a minor issue but I don't know a better way than to open a formal issue. So anyone can find this and know that solving this specific topic is ongoing.

pfeerick, 2025-08-14:
The matter of Companion showing more protocols than the currently running
MPM firmware supports will probably be caught up in the runtime settings
TODO for Companion (still some way away as it was only at the early concept
stage last time it was discussed) but updating the Companion (and firmware
no-module) protocols list is underway.

elecpower, 2025-09-13:
TODO reference above is #5469

closed 2026-01-06 by pfeerick