EdgeTX/edgetx #6519: During yesterday's Companion installation, the program encountered an error.

koaxheli, 2025-08-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

The dll file VCRUNTIME140_1.dll cannot be found.

[image]

### Expected Behavior

All previous updates have always worked without any problems.

### Steps To Reproduce

The error occurs with yesterday's version companion-windows-3.0.0.exe.
The previous version is still OK.

### Version

2.11.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

elecpower, 2025-09-19:
From testing seems we also need to update the installed version too if older than expected by Companion and model simulators.

Struck issues where the simulators launched from Companion crash on loading or closing. Windows Event log entries reference windows and Qt dlls.

Updated using vc_redist.x64.exe and crashes have gone away for the time being.

raphaelcoeffic, 2025-09-24:
This is rather strange as this is supposed to be shipped with Windows 10 and Windows 11. So which windows version is impacted here? Do we really want to ship this on top just because some old windows version doesn’t ship the C Runtime which is mandatory for each application? (More or less)

elecpower, 2025-09-24:
Just built a win11 pro vm from official 24H2 iso. Installed Companion and on launch MSVCP140.DLL VCRUNTIME140.DLL and VCRUNTIME140_1.DLL not found.

There are copies of these files scattered around the Windows subfolders but I can only assume they are localised.