EdgeTX/edgetx #6531: Mute if no sound option for FrSky X9D+

shyney7, 2025-08-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Please add the option to mute the audio if no sound is being played also for the FrSky X9D+.
It seems that this option is missing on the latest Firmware for this radio (2.11.2)

### Describe the solution you'd like

Option in the Hardware Tab to mute the sound if no audio is played.

### Additional context

https://github.com/EdgeTX/edgetx/issues/4595

https://manual.edgetx.org/color-radios/radio-settings/hardware

shyney7, 2025-08-19:
Maybe I'm thinking wrong but you can control the volume on the X9D so it should be possible to mute the volume by lowering the volume to zero?

pfeerick, 2025-08-19:
No, this is is different. Being able to mute the audio amp means you can shuts down the audio circuit output entirely. It is not just setting it to 0, as that would mean leaving the amp and thus audio circuit active, resulting in RF and electrical noise still able to come through the speaker.

3djc, 2025-08-20:
Yes exactly, hardware doesn't support it

closed 2025-08-20 by shyney7