EdgeTX/edgetx #6554: V3 W64 Companion/Simulator: various issues

mha1, 2025-08-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Log File Viewer: 
- after loading a longer log file changing sensors to display takes very long. try attached log file and compare to V2.11
- wrong time stamps, see attached image and compare to V2.11

[image]

[PikePerf-2025-08-16-172434.csv](https://github.com/user-attachments/files/22004416/PikePerf-2025-08-16-172434.csv)

Simulator:
- Simulator doesn't show Model Notes. with Display Checklist activated

### Expected Behavior

V2.11 behavior:
- fast display of log view graphs after changing sensors
- show correct time stamps 
- simulator shows model notes if Display Checklist is activated

### Steps To Reproduce

Use attached log file and compare 2.11 to V3
Have a model with model notes .txt and display checklist activated in simulator

### Version

Other (Please specify below)

### Transmitter

Jumper T15

### Operating System (OS)

Windows

### OS Version

W11

### Anything else?

[image]

raphaelcoeffic, 2025-08-27:
> Simulator doesn't show Model Notes. with Display Checklist activated

Regarding this: is the same behaviour to be seen in the `nightly` firmware, or is it really specific to the simulator? Also, are you sure that the notes are really on the SD card image linked?

koaxheli, 2025-08-28:
I opened your file with EdgeTX v 3.0.0 in Companion and found no errors.

[image]

koaxheli, 2025-08-28:
[image]

mha1, 2025-08-28:
> I opened your file with EdgeTX v 3.0.0 in Companion and found no errors.

Windows? Which CPN version exactly?

mha1, 2025-08-28:
> [@mha1](https://github.com/mha1) did you use your compilation or a nightly ?

Nooo, of course it's an official build. It's the latest main build I could find: https://github.com/EdgeTX/edgetx/actions/runs/17084001814

[image]

mha1, 2025-08-28:
> Latest version v3.0.0 ([1f34bf1](https://github.com/EdgeTX/edgetx/commit/1f34bf126bf8651b0621e9bed6efd8701d33262e))

This is ancient. And it looks like a Linux or Mac build.

[image]

mha1, 2025-09-03:
> This is Windows, not Mac or Linux.

Ok, good. Does it show time stamps and how's switching between available fields?

I installed the same version (W64 GH build Commit 7230b26d, https://github.com/EdgeTX/edgetx/actions/runs/17367481937) and it has the same issues I initially reported.

koaxheli, 2025-09-04:
To switch between available fields, simply click on the field you want to see.

[image]

koaxheli, 2025-09-04:
[image]
Display also works on the transmitter.

mha1, 2025-09-04:
@pfeerick you are in tz GMT+10, aren't you. In in GMT+1. Good thinking!

I can confirm the missing/empty model notes too.