EdgeTX/edgetx #6565: Reset Flight can cause crash

ror1948, 2025-08-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have many things reset when using reset switch(SH) on TX16s.
However if you include a "reset Flight" and your switches are NOT in the normal power on condition then the RF module is NOT turned on and If this is done while flying, your plane is flying with no control.

This can be set at the SYS level or the MDL level.

Solution: Do not check state of switches during reset flight.

### Expected Behavior

Reset flight does not check state of switches and RF signal is maintained.

### Steps To Reproduce

create global function that executes "reset flight" when SH is UP
if switches are not in expected position then the LED will go RED and you will be prompted to set switches to desired position. RED LED means RF signal is OFF and Rx has no signal.

### Version

2.11.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[SC.bmp](https://github.com/user-attachments/files/22069105/SC.bmp)

pfeerick, 2025-09-01:
> Solution: Do not check state of switches during reset flight.

No, most definitely not. If it doesn't already, it should probably behave exactly the same as startup or switching models... where you get the fullscreen notification/audio prompt that model/telemetry is still connected - to which if you confirm you want to continue, you are accepting the link will be lost. 

To do what you are suggesting will need to do more than just skip checking switches, you need all preflight checks skipped, and for the RF module to not be power cycled at all (otherwise you will still get a disconnect, and who knows how long before reconnect as that varies by both RF module and link/receiver.

ror1948, 2025-09-01:
why do you want a switch action available that can kill your RF link?
I know of no other radio system that has that.
When I am flying I am not looking at the screen.

raphaelcoeffic, 2025-09-01:
> we need to rename "Flight" SF reset choice to "Session"

+1

pfeerick, 2025-09-01:
RF would *usually* only be off in combination with error state (i.e. red LED) if you have switched models (because the RF and mixer are shutdown between model unload/load) or are startup up the radio (because the RF and mixer haven't been startup up yet). Otherwise, both will usually be running, but obviously there is still some sort of error needing your attention. 

Maybe this needs to be explained [in the manual](https://manual.edgetx.org/) somehow? Any thoughts on where it might fit? What about under the User Interface sections for Colour LCD and B&W? i.e. add a section about standard behaviours/conventions?

Either way, thanks for making me realise a few other texts need updating ;)

ror1948, 2025-09-04:
Just a thought:
-use red LED for RF off Green LED for RF on ONLY.
-no need to have LED red when there is an error because we have a screen telling us what the error is and once you press a key, RED LED is gone and so is the screen.